espressif/esp-apple-homekit-adk #1: Bluetooth support

langovoi, 2020-05-22:
Do you have plan to make bluetooth support?

shahpiyushv, 2020-06-29:
@langovoi , we do have plans to add it, but not any time soon. We are open to contributions though :)

HomeKidd, 2020-07-01:
> @langovoi , we do have plans to add it, but not any time soon. We are open to contributions though :)

BLE support with deepsleep would be great! 😄

dakhnod, 2020-09-07:
So, did anything happen?

I can see a "#if (BLE)", but will defining BLE suffice in order to enable BLE support?

HomeKidd, 2020-09-07:
@dakhnod i dont think so, last commit was at May 14. That definition for BLE is just part of HomeKit-ADK😉

shahpiyushv, 2020-09-07:
@dakhnod , what @HomeKidd said is (unfortunately) correct.

crasmussen, 2020-09-22:
Hi @shahpiyushv , any estimate on when BLE might be supported?

mriksman, 2020-12-13:
Super bummed to find that setting `define BLE 1` did not compile.
```
../main/app_main.c:130:17: error: 'HAPPlatformBLEPeripheralManagerOptions' {aka 'struct <anonymous>'} has no member named 'keyValueStore'
     blePMOptions.keyValueStore = &platform.keyValueStore;
```

+1 for BLE support.

yrjyrj123, 2022-02-09:
Is there any update?