espressif/esp-apple-homekit-adk #2: Andino IDE suport

Asteliks, 2020-06-06:
Could you make a version of this compatible with the Arduino IDE?

jwktje, 2020-06-25:
I would also love this. But I think it’s probably unlikely to happen.

shahpiyushv, 2020-06-25:
It isn't that straight forward because a majority of code and APIs is from the Apple ADK directly. The Arduino support for ESP32 is mostly in the form of pre-built libraries and a wrappers on top of that, which could be quite cumbersome (but not impossible) for this project. I do not plan to do this myself any time soon, but contributions from the community are always welcome :)

jwktje, 2020-06-25:
Yeah that's what I figured. After looking through the Lightbulb example though, I gotta be honest; it looks like a lot. 
I've used maximkulkin/esp-homekit for a couple DIY homekit devices around the house. It took some time to figure out the workflow but after getting past the initial learning curve the final code seemed reasonable in size.

I'm sure that this esp-apple-homekit-adk would work great once you take the time to master it. But as a hobbyist myself it feels like the barrier to entry is quite high. The code looks daunting to me (but probably not to actual embedded developers).

The reason that Arduino IDE support would be cool is not really because I would like to use Arduino. I guess the question (for me) boils down to; I wish it was as simple as Arduino to a newcomer. 

If I imagine an arduino sketch that needs to connect to Wifi and would allow the on/off states of a GPIO (for something like a relay for example) through a HTTP server endpoint, I could make that work in less than 50 LOC with some libraries. 

Obviously going through HomeKit for something like this adds complexity. And obviously Arduino libraries keep the sketch code short (but the compiled binary larger). And of course the IDF is way more "close to the metal", making for (on average) more includes at the top and more verbose code to get something running (with the added benefit of less dependencies, better performance and more freedom to make the firmware work exactly how you want without it being dictated by a library).

I totally understand all that. After hearing that the Homekit ADK was open source and supported by Espressif directly I was super stoked. But I've already had 3 different days where I had some free time wanting to get started with it, but ended up abandoning it. Mostly because there isn't much to tinker with. 

Maybe someone can make a library on top of this esp-apple-homekit-adk port that would handle most of the universal HomeKit parts?

vicfrancco, 2020-08-03:
I understand what jwktje has said. For Newbies like me, it’s a very high learning curve to get to know how to program in the ESP-IDF way and to match it with Apple ADK way.

Furthermore, I think the “monster” could be much shorter if we had examples that dealt with our world (makers world I mean).

After reading the Lightbulb example here and comparing to the Apple’s ADK one, what I’ve understood is that the Espressif’s example is a port of Apple’s example.

I think that if Espressif could bring us an example that shows us how to set the GPIO32 (or other GPIO) level (high or low) of an ESP32 that controls a relay to which that Lightbulb PF the example, it would be much more useful.

bfaliszek, 2020-08-07:
I'm also looking for a Arduino version. I found recently - https://github.com/Mixiaoxiao/Arduino-HomeKit-ESP32 and the example works. It is a pity that the description says that the library will no longer be developed.

vicfrancco, 2020-08-16:
> 
> 
> I'm also looking for a Arduino version. I found recently - https://github.com/Mixiaoxiao/Arduino-HomeKit-ESP32 and the example works. It is a pity that the description says that the library will no longer be developed.

Yes, the example works for me too. Within 5 minutes reading the code I was able to find out how he changes the level of GPIO 2.

Unfortunately, seems like the library will no longer be developed, because Espressif released the official Homekit ADK, witch, up-to-date, seems to be not maker's friendly at all.

A well documented example of the simple led approach woud be very important.

bfaliszek, 2020-08-17:
I found one more library to handle home kit on esp32. It seems significant and more stable to me than the previous one: https://github.com/Yurik72/ESPHap/tree/master

Asteliks, 2020-08-25:
I have managed to make some examples using this library. Actually ones everything starts working it's not that hard. So far it is stable. Link to my repo: [https://github.com/Asteliks/AppleHomeKitADK-ESP32](https://github.com/Asteliks/AppleHomeKitADK-ESP32)

JohnWickham, 2020-09-22:
Wanted to +1 @bfaliszek’s recommendation of https://github.com/Yurik72/ESPHap. It’s the first HAP library that I’ve managed to get working with Arduino and it works nearly perfectly.

Before that I was using https://github.com/maximkulkin/esp-homekit, but it was cumbersome and required Linux virtualization as Espressif’s 8266 toolchain no longer builds on macOS Catalina.

shahpiyushv, 2020-09-25:
We have now opened up our own implementation of HomeKit. Check out the ESP HomeKit SDK [here](https://github.com/espressif/esp-homekit-sdk) and see if suits your requirements.

JohnWickham, 2020-09-25:
Thanks @shahpiyushv, that looks interesting. But it seems that it only supports ESP32, not ESP8266, correct?

> You can use esp-homekit-sdk with any ESP32 or ESP32-S2 board

shahpiyushv, 2020-09-25:
> Thanks @shahpiyushv, that looks interesting. But it seems that it only supports ESP32, not ESP8266, correct?
> 
> > You can use esp-homekit-sdk with any ESP32 or ESP32-S2 board

@JohnWickham , actually, with some changes, it did indeed work fine with ESP8266. I will make the required changes in a few days.

arihantdaga, 2020-09-27:
@shahpiyushv Does this mean that now it'll be Arduino compatible? What would one need to do to in order to use the sdk in an arduino project ?

shahpiyushv, 2020-09-28:
@arihantdaga , Arduino support may still not be available anytime soon, but this new SDK has a different set of (simpler) APIs for writing an accessory with many more examples and ready to use functions for different services. It also has other integrations like use of the QR  code based Wi-Fi provisioning and also gels well with our other products, particularly [ESP RainMaker](https://rainmaker.espressif.com/). We already have an [example](https://github.com/espressif/esp-rainmaker/tree/master/examples/homekit_switch) available which will give you Alexa + Google + Siri control. Since this new SDK is developed in-house, it may probably be easier for us to add into Arduino in future.

Yurik72, 2020-09-29:
@shahpiyushv, could please comment realistic scenario for Arduino.  
your opened repository https://github.com/espressif/esp-homekit-sdk  looks very promise. I'd say,  the efforts to support arduino for ESP32 is quite clear at least to me. But to support ESP8266, based on this code, we need  Arduino for ESP8266 based on the FreeRTOS implementation . Those discussed a lot of times, Therefore a question, what is your  realitys and plans ?
This question is only about that usage of ESP-IDF  is not a problem for "experienced" users,l but for guys like schoolboys is not possible.

shahpiyushv, 2020-09-30:
> @shahpiyushv, could please comment realistic scenario for Arduino.
> your opened repository https://github.com/espressif/esp-homekit-sdk looks very promise. I'd say, the efforts to support arduino for [...]

The esp-homekit-sdk would work with esp-idf and ESP8266-RTOS-SDK. With regards to Arduino, we may add it for ESP32, but not ESP8266. I guess ESP32 + Arduino should still be good enough for the "schoolboys" :)

shahpiyushv, 2020-10-08:
We have now added support for ESP8266 in [esp-homekit-sdk]( https://github.com/espressif/esp-homekit-sdk ).

tomdmt, 2021-07-21:
Any news on that Arduino IDE support?

shahpiyushv, 2021-07-22:
@tomdmt , sorry, but we aren't really working on adding this into Arduino :-/

FloBer431, 2024-01-24:
@shahpiyushv Is there anything new about this topic? Almost 3 years now, is there still no Arduino IDE support for ESP Apple Homekit?