espressif/esp-apple-homekit-adk #3: mbedtls error

rash27, 2020-06-08:
Hi, 

I am seeing the below error when I try flashing to ESP32:

Generated /Users/rashmi/esp-apple-homekit-adk/examples/Lightbulb/build/bootloader/bootloader.bin
[661/1035] Building C object esp-idf/port/CMakeFiles/__idf_port.dir/src/HAPPlatformLog.c.obj
/Users/rashmi/esp-apple-homekit-adk/port/src/HAPPlatformLog.c: In function 'HAPPlatformLogPOSIXError':
/Users/rashmi/esp-apple-homekit-adk/port/src/HAPPlatformLog.c:60:13: warning: initialization of 'int' from 'char *' makes integer from pointer without a cast [-Wint-conversion]
     int e = strerror_r(errorNumber, errorString, sizeof errorString);
             ^~~~~~~~~~
[708/1035] Building C object esp-idf/port/CMakeFiles/__idf_port.dir/Users/rashmi/esp-apple-homekit-adk/homekit_adk/PAL/Crypto/MbedTLS/HAPMbedTLS.c.obj
FAILED: esp-idf/port/CMakeFiles/__idf_port.dir/Users/rashmi/esp-apple-homekit-adk/homekit_adk/PAL/Crypto/MbedTLS/HAPMbedTLS.c.obj
/Users/rashmi/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -DHAP_LOG_LEVEL=1 -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -Iconfig -I/Users/rashmi/esp-apple-homekit-adk/port/include -I/Users/rashmi/esp-apple-homekit-adk/homekit_adk/HAP -I/Users/rashmi/esp-apple-homekit-adk/homekit_adk/PAL -I/Users/rashmi/esp-apple-homekit-adk/homekit_adk/External/Base64 -I/Users/rashmi/esp-apple-homekit-adk/homekit_adk/External/JSON -I/Users/rashmi/esp-apple-homekit-adk/homekit_adk/External/HTTP -I/Users/rashmi/esp-idf/components/newlib/platform_include -I/Users/rashmi/esp-idf/components/freertos/include -I/Users/rashmi/esp-idf/components/freertos/xtensa/include -I/Users/rashmi/esp-idf/components/heap/include -I/Users/rashmi/esp-idf/components/log/include -I/Users/rashmi/esp-idf/components/lwip/include/apps -I/Users/rashmi/esp-idf/components/lwip/include/apps/sntp -I/Users/rashmi/esp-idf/components/lwip/lwip/src/include -I/Users/rashmi/esp-idf/components/lwip/port/esp32/include -I/Users/rashmi/esp-idf/components/lwip/port/esp32/include/arch -I/Users/rashmi/esp-idf/components/soc/src/esp32/. -I/Users/rashmi/esp-idf/components/soc/src/esp32/include -I/Users/rashmi/esp-idf/components/soc/include -I/Users/rashmi/esp-idf/components/esp_rom/include -I/Users/rashmi/esp-idf/components/esp_common/include -I/Users/rashmi/esp-idf/components/esp_system/include -I/Users/rashmi/esp-idf/components/xtensa/include -I/Users/rashmi/esp-idf/components/xtensa/esp32/include -I/Users/rashmi/esp-idf/components/esp32/include -I/Users/rashmi/esp-idf/components/driver/include -I/Users/rashmi/esp-idf/components/driver/esp32/include -I/Users/rashmi/esp-idf/components/esp_ringbuf/include -I/Users/rashmi/esp-idf/components/efuse/include -I/Users/rashmi/esp-idf/components/efuse/esp32/include -I/Users/rashmi/esp-idf/components/espcoredump/include -I/Users/rashmi/esp-idf/components/esp_timer/include -I/Users/rashmi/esp-idf/components/soc/soc/esp32/. -I/Users/rashmi/esp-idf/components/soc/soc/esp32/include -I/Users/rashmi/esp-idf/components/soc/soc/esp32/../include -I/Users/rashmi/esp-idf/components/vfs/include -I/Users/rashmi/esp-idf/components/esp_wifi/include -I/Users/rashmi/esp-idf/components/esp_wifi/esp32/include -I/Users/rashmi/esp-idf/components/esp_event/include -I/Users/rashmi/esp-idf/components/esp_netif/include -I/Users/rashmi/esp-idf/components/esp_eth/include -I/Users/rashmi/esp-idf/components/tcpip_adapter/include -I/Users/rashmi/esp-idf/components/app_trace/include -I/Users/rashmi/esp-idf/components/nvs_flash/include -I/Users/rashmi/esp-idf/components/spi_flash/include -I/Users/rashmi/esp-idf/components/mbedtls/port/include -I/Users/rashmi/esp-idf/components/mbedtls/mbedtls/include -I/Users/rashmi/esp-idf/components/mbedtls/esp_crt_bundle/include -I/Users/rashmi/esp-idf/components/mdns/include -I/Users/rashmi/esp-idf/components/console -mlongcalls -Wno-frame-address   -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.2-dev-1660-g7d7521367-dirty\" -DESP_PLATFORM -MD -MT esp-idf/port/CMakeFiles/__idf_port.dir/Users/rashmi/esp-apple-homekit-adk/homekit_adk/PAL/Crypto/MbedTLS/HAPMbedTLS.c.obj -MF esp-idf/port/CMakeFiles/__idf_port.dir/Users/rashmi/esp-apple-homekit-adk/homekit_adk/PAL/Crypto/MbedTLS/HAPMbedTLS.c.obj.d -o esp-idf/port/CMakeFiles/__idf_port.dir/Users/rashmi/esp-apple-homekit-adk/homekit_adk/PAL/Crypto/MbedTLS/HAPMbedTLS.c.obj   -c /Users/rashmi/esp-apple-homekit-adk/homekit_adk/PAL/Crypto/MbedTLS/HAPMbedTLS.c
/Users/rashmi/esp-apple-homekit-adk/homekit_adk/PAL/Crypto/MbedTLS/HAPMbedTLS.c: In function 'HAP_X25519_scalarmult':
/Users/rashmi/esp-apple-homekit-adk/homekit_adk/PAL/Crypto/MbedTLS/HAPMbedTLS.c:119:19: error: implicit declaration of function 'mbedtls_ecp_read_key'; did you mean 'mbedtls_ecp_gen_key'? [-Werror=implicit-function-declaration]
         int ret = mbedtls_ecp_read_key(MBEDTLS_ECP_DP_CURVE25519, &our_key, n, X25519_SCALAR_BYTES);
                   ^~~~~~~~~~~~~~~~~~~~
/Users/rashmi/esp-apple-homekit-adk/homekit_adk/PAL/Crypto/MbedTLS/HAPMbedTLS.c:107:9: note: in definition of macro 'WITH'
         X; \
         ^
/Users/rashmi/esp-apple-homekit-adk/homekit_adk/PAL/Crypto/MbedTLS/HAPMbedTLS.c:118:5: note: in expansion of macro 'WITH_ECP_KEYPAIR'
     WITH_ECP_KEYPAIR(our_key, {
     ^~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
[721/1035] Building C object esp-idf/port/CMakeFiles/__idf_port.dir/Users/rashmi/esp-apple-homekit-adk/homekit_adk/HAP/HAPIPAccessory.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

Any clue why this is happening? I am following the exact same instructions you have on your page with git checkout -b mbedtls-2.16.6-adk origin/mbedtls-2.16.6-adk

Thanks for your help

shahpiyushv, 2020-06-09:
@rash27  Can you go to $IDF_PATH/components/mbedtls/mbedtls and execute `git log -1`? You should see this commit:

```
commit 61aa5edb1c68ecc6324e98e57628582d8861e87b 
Author: Janos Follath <janos.follath@arm.com>
Date:   Tue Feb 26 12:36:52 2019 +0000

    Make ecp_get_type public
    
    The ecp_get_type function comes handy in higher level modules and tests
    as well. It is not inline anymore, to enable alternative implementations
    to implement it for themselves.
```

rash27, 2020-06-10:
Thanks! I was able to flash it successfully. 

When I connect to Home app, I see issues turning the light on or off. It doesnt update the status. 

1970-01-01'T'00:00:00'Z'	Default	[com.apple.mfi.HomeKit.Platform:KeyValueStore] keyValueStore nvs Initialised
1970-01-01'T'00:00:00'Z'	Default	[com.apple.mfi.HomeKit.Platform:KeyValueStore] keyValueStore fctry Initialised
I (856) wifi:wifi driver task: 3ffcd544, prio:23, stack:6656, core=0
I (856) system_api: Base MAC address is not set
I (856) system_api: read default base MAC address from EFUSE
I (886) wifi:wifi firmware version: 517093f
I (886) wifi:wifi certification version: v7.0
I (886) wifi:config NVS flash: enabled
I (886) wifi:config nano formating: disabled
I (886) wifi:Init dynamic tx buffer num: 32
I (886) wifi:Init data frame dynamic rx buffer num: 32
I (896) wifi:Init management frame dynamic rx buffer num: 32
I (896) wifi:Init management short buffer num: 32
I (906) wifi:Init static rx buffer size: 1600
I (906) wifi:Init static rx buffer num: 10
I (916) wifi:Init dynamic rx buffer num: 32
1970-01-01'T'00:00:00'Z'	Default	[com.apple.mfi.HomeKit.Platform:KeyValueStore] Error (4354). Key 20 not found in KeyStore
1970-01-01'T'00:00:00'Z'	Default	[com.apple.mfi.HomeKit.Platform:KeyValueStore] Error (4354). Key 21 not found in KeyStore
I (1046) phy: phy_version: 4181, c9a03e2, Mar 10 2020, 12:45:06, 0, 0
I (1046) wifi:mode : sta (24:6f:28:0b:7b:bc)
I (1056) wifi station: wifi_init_sta finished.
I (1176) wifi:new:<3,0>, old:<1,0>, ap:<255,255>, sta:<3,0>, prof:1
I (1176) wifi:state: init -> auth (b0)
I (1186) wifi:state: auth -> assoc (0)
I (1196) wifi:state: assoc -> run (10)
I (1236) wifi:connected with Jacaranda, aid = 17, channel 3, BW20, bssid = 3c:37:86:79:c8:e0
I (1236) wifi:security type: 3, phy: bgn, rssi: -53
I (1236) wifi:pm start, type: 1

I (1326) wifi:AP's beacon interval = 204800 us, DTIM period = 2

It shows the AP and the password correctly

1970-01-01'T'00:00:02'Z'	Default	[com.apple.mfi.HomeKit.Core:AccessoryServer] Version information:
libhap: Unknown
  - Version: Internal (Jun  9 2020 22:31:06) - compatibility version 7
Using platform: ESP32
  - Version: v4.2-dev-1660-g7d7521367-dirty (Jun  9 2020 22:31:04) - compatibility version 7
  - Available features:
    - Key-Value store
    - Accessory setup manager
    - Service discovery
1970-01-01'T'00:00:02'Z'	Default	[com.apple.mfi.HomeKit.Platform:Clock] Using 'gettimeofday'.

shahpiyushv, 2020-06-10:
@rash27 can you tell how exactly are you testing this? Did pairing from the Home app work fine?

rash27, 2020-06-11:
Yes, the pairing worked fine. But the status doesn't change from OFF to ON state.

shahpiyushv, 2020-06-11:
@rash27 , if you are just checking for the on/off prints, you may probably not see any. You can confirm by adding some print statement in the [HandleLightBulbOnWrite](https://github.com/espressif/esp-apple-homekit-adk/blob/master/examples/Lightbulb/main/App.c#L163) function or by changing the log level to "Info" by using idf.py menuconfig -> HomeKit -> HAP Log Level -> Info

rash27, 2020-06-12:
I meant the ON or OFF state on the Home IOS app. When you click on the Acme Light bulb (which I am able to see and add the accessory to my home app), it doesn't change the state of the bulb to ON.

shahpiyushv, 2020-06-22:
@rash27 , did you try adding the print like I had mentioned above?

shahpiyushv, 2020-10-11:
Closing this issue as it has diverged and has no activity for past many days.

closed 2020-10-11 by shahpiyushv