espressif/esp-apple-homekit-adk #4: Change device state manually

agtabesh, 2020-06-19:
Hi there,

Recently, I used `Lightbulb` example in one of my projects on `ESP32` and everything is working, as expected, perfectly. 

The problem we came up with was changing the state of the light manually, using a switch or something else but we haven't done it yet!

Actually we changed `accessoryConfiguration.state.lightBulbOn` to what value we want but it doesn't broadcast and update the status of the light in the app on my iPhone.

I believe that there must be a way (such as an API) to change device state and let other devices (including my iPhone) know of change, but we don't know how!

If someone have had the same need and experience, I'll be so appreciated to share.

Best regards.

shahpiyushv, 2020-06-29:
@agtabesh HAPAccessoryServerRaiseEvent() is probably the function to be called. You can check the usage here: https://github.com/espressif/esp-apple-homekit-adk/blob/master/examples/Lightbulb/main/App.c#L163

aselafernando, 2020-07-22:
@agtabesh Did you ever solve this?
The example linked shows how you force a characteristic update from within a characteristic using:

HAPAccessoryServerRaiseEvent(server, request->characteristic, request->service, request->accessory);

However this function does not do anything if called outside of a characteristic callback. For example if you had a light switch that could be triggered manually (physically turning it on) or via homekit. If triggered manually you will want the status to flow back to homekit. I have not found a way to do this successfully.

HAPAccessoryServerRaiseEvent(accessoryConfiguration.server, &characteristicThatUpdated, &serviceThatCharacteristicBelongsTo, &accessory)

Even though I don't get any error, HomeKit does not get the updated status

odx, 2020-08-20:
I verified that I use the very same objects (pointer addresses) to call this function in my own tasks compared to what happens in the ADK callbacks.
I feel like this is a threading issue as the state gets updated under unpredictable circumstances for me. 

Any idea how to execute code within the HAP runloop?

shahpiyushv, 2020-08-21:
@odx Have you checked [HAPPlatformRunLoopScheduleCallback()](https://github.com/apple/HomeKitADK/blob/5c19c9eb2bff41115ec8f1c763c71bd6a69545e5/PAL/HAPPlatformRunLoop.h#L46)?

skanico, 2020-09-04:
@odx I'm running in the same issue. Could you tell me how you used this function to trigger homekit update?
Thanks

odx, 2020-09-04:
You will have to implement a callback function that can be added to the runloop which satisfies this type:
`typedef void (*HAPPlatformRunLoopCallback)(void* _Nullable context, size_t contextSize);`
Within this function you may raise the HAP event which will in turn query the read functions for the characteristic.
in my case the function looks like this:
```
void updateHAPCurrentState(void* _Nullable context, size_t contextSize)
{
    HAPAccessoryServerRaiseEvent(accessoryConfiguration.server, &lockMechanismLockCurrentStateCharacteristic, &lockMechanismService, AppGetAccessoryInfo());
}
```

You can trigger an execution of your function by adding it to the runloop:
HAPError err = HAPPlatformRunLoopScheduleCallback(updateHAPCurrentState, NULL, 0);

If you need to pass data you may use the context pointer and contextSize.

Keep in mind that the callback will be invoked some time later when the runloop kicks in.

skanico, 2020-09-04:
Wow, many thanks! I couldn't have done it without you.
Some more fine tuning, but it works!

shahpiyushv, 2020-10-11:
Closing this issue since it has been resolved.

closed 2020-10-11 by shahpiyushv