espressif/esp-apple-homekit-adk #5: ESP32 "Fault	assertion failed" message and reboots

AramVartanyan, 2020-06-28:
I have tried unchanged version of the Lightbulb example, following the building instructions.
It works fine, however I have noticed that it keeps rebooting around 10 minutes after closing the session with the "iOS" device.
I really don't understand what is causing the issue, but I want to help and mention it.
This is a part of the screen log (I hope it is enough):

```
1970-01-01'T'00:10:41'Z'	Info	[com.apple.mfi.HomeKit.Core:PairingPairVerify] Pair Verify procedure completed (pairing ID 0).
1970-01-01'T'00:10:41'Z'	Debug	[com.apple.mfi.HomeKit.Core:IPAccessoryServer] session:0x3ffb9cd0:<
    0000  48545450 2f312e31 20323030 204f4b0d 0a436f6e 74656e74 2d547970 653a2061    HTTP/1.1 200 OK..Content-Type: a
    0020  70706c69 63617469 6f6e2f70 61697269 6e672b74 6c76380d 0a436f6e 74656e74    pplication/pairing+tlv8..Content
    0040  2d4c656e 6774683a 20330d0a 0d0a0601 04                                     -Length: 3.......
1970-01-01'T'00:10:41'Z'	Debug	[com.apple.mfi.HomeKit.Core:IPAccessoryServer] session:0x3ffb9cd0:input closed
1970-01-01'T'00:10:41'Z'	Debug	[com.apple.mfi.HomeKit.Core:IPAccessoryServer] session:0x3ffb9cd0:closing
1970-01-01'T'00:10:41'Z'	Debug	[com.apple.mfi.HomeKit.Core:IPAccessoryServer] session:0x3ffb9cd0:closing security context
1970-01-01'T'00:10:41'Z'	Debug	[com.apple.mfi.HomeKit.Core:IPAccessoryServer] Closing HAP session.
1970-01-01'T'00:10:41'Z'	Debug	[com.apple.mfi.HomeKit.Core:Session] HAPSessionRelease
1970-01-01'T'00:10:41'Z'	Debug	[com.apple.mfi.HomeKit.Core:Session] HAPSessionInvalidate
1970-01-01'T'00:10:41'Z'	Debug	[com.apple.mfi.HomeKit.Core:IPAccessoryServer] session:0x3ffb9cd0:closing TCP stream
1970-01-01'T'00:10:41'Z'	Debug	[com.apple.mfi.HomeKit.Platform:TCPStreamManager] shutdown(51, SHUT_RDWR);
1970-01-01'T'00:10:41'Z'	Fault	assertion failed - HAPPlatformLogPOSIXError @ /Users/Logic/esp32_test/esp-apple-homekit-adk/port/src/HAPPlatformLog.c:65

abort() was called at PC 0x400d5fa3 on core 0
0x400d5fa3: syscall_not_implemented_aborts at /Users/Logic/esp32_test/esp-idf/components/newlib/syscalls.c:39

Backtrace:0x4008760f:0x3ffc9540 0x40087cad:0x3ffc9560 0x4008e8fe:0x3ffc9580 0x400d5fa3:0x3ffc95f0 0x40163db2:0x3ffc9610 0x400e9a65:0x3ffc9630 0x400d8b74:0x3ffc9650 0x400e9ca6:0x3ffc9670 0x400d87a2:0x3ffc97a0 0x400e320a:0x3ffc97c0 0x400e32bf:0x3ffc9800 0x400e4ba5:0x3ffc9820 0x400e5aa9:0x3ffc9850 0x400d7e51:0x3ffc9870 0x400d721d:0x3ffc9890 0x400d7a20:0x3ffc98b0 0x400d6487:0x3ffc98f0 0x40087cb5:0x3ffc9910
0x4008760f: panic_abort at /Users/Logic/esp32_test/esp-idf/components/esp_system/panic.c:330

0x40087cad: esp_system_abort at /Users/Logic/esp32_test/esp-idf/components/esp_system/system_api.c:100

0x4008e8fe: abort at /Users/Logic/esp32_test/esp-idf/components/newlib/abort.c:46

0x400d5fa3: syscall_not_implemented_aborts at /Users/Logic/esp32_test/esp-idf/components/newlib/syscalls.c:39

0x40163db2: exit at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/exit.c:64

0x400e9a65: HAPPlatformAbort at /Users/Logic/esp32_test/esp-apple-homekit-adk/port/src/HAPPlatformAbort.c:27

0x400d8b74: HAPAssertInternal at /Users/Logic/esp32_test/esp-apple-homekit-adk/homekit_adk/PAL/HAPAssert.c:20

0x400e9ca6: HAPPlatformLogPOSIXError at /Users/Logic/esp32_test/esp-apple-homekit-adk/port/src/HAPPlatformLog.c:65 (discriminator 2)

0x400d87a2: HAPPlatformTCPStreamClose at /Users/Logic/esp32_test/esp-apple-homekit-adk/port/src/HAPPlatformTCPStreamManager.c:516

0x400e320a: CloseSession at /Users/Logic/esp32_test/esp-apple-homekit-adk/homekit_adk/HAP/HAPIPAccessoryServer.c:513 (discriminator 1)

0x400e32bf: handle_input_closed at /Users/Logic/esp32_test/esp-apple-homekit-adk/homekit_adk/HAP/HAPIPAccessoryServer.c:3553

0x400e4ba5: ReadInboundData at /Users/Logic/esp32_test/esp-apple-homekit-adk/homekit_adk/HAP/HAPIPAccessoryServer.c:3595

0x400e5aa9: HandleTCPStreamEvent at /Users/Logic/esp32_test/esp-apple-homekit-adk/homekit_adk/HAP/HAPIPAccessoryServer.c:3622

0x400d7e51: HandleTCPStreamFileHandleCallback at /Users/Logic/esp32_test/esp-apple-homekit-adk/port/src/HAPPlatformTCPStreamManager.c:731

0x400d721d: ProcessSelectedFileHandles at /Users/Logic/esp32_test/esp-apple-homekit-adk/port/src/HAPPlatformRunLoop.c:303

0x400d7a20: HAPPlatformRunLoopRun at /Users/Logic/esp32_test/esp-apple-homekit-adk/port/src/HAPPlatformRunLoop.c:640

0x400d6487: main_task at /Users/Logic/esp32_test/esp-apple-homekit-adk/examples/Lightbulb/build/../main/app_main.c:336

0x40087cb5: vPortTaskWrapper at /Users/Logic/esp32_test/esp-idf/components/freertos/xtensa/port.c:143
```

shahpiyushv, 2020-06-29:
@AramVartanyan , can you see what happens if you comment out these shutdown calls and the error check code around them?
https://github.com/espressif/esp-apple-homekit-adk/blob/master/port/src/HAPPlatformTCPStreamManager.c#L512,
https://github.com/espressif/esp-apple-homekit-adk/blob/master/port/src/HAPPlatformTCPStreamManager.c#L328

AramVartanyan, 2020-06-29:
Thank you for your fast respond.
I have commented the two "Shutdowns" you suggested. It doesn't restart anymore. Which is very nice! :)

The output is as follows:

```
1970-01-01'T'00:34:37'Z'	Debug	[com.apple.mfi.HomeKit.Core:PairingPairVerify] <private> Pair Verify M3: iOSDeviceInfo.
1970-01-01'T'00:34:37'Z'	Debug	[com.apple.mfi.HomeKit.Core:PairingPairVerify] <private> Pair Verify M3: kTLVType_Signature.
1970-01-01'T'00:34:37'Z'	Debug	[com.apple.mfi.HomeKit.Core:PairingPairVerify] Pair Verify M4: Verify Finish Response.
1970-01-01'T'00:34:37'Z'	Debug	[com.apple.mfi.HomeKit.Core:PairingPairVerify] <private> Pair Verify Start Session: AccessoryToControllerKey
1970-01-01'T'00:34:37'Z'	Debug	[com.apple.mfi.HomeKit.Core:PairingPairVerify] <private> Pair Verify Start Session: ControllerToAccessoryKey
1970-01-01'T'00:34:37'Z'	Info	[com.apple.mfi.HomeKit.Core:PairingPairVerify] Pair Verify procedure completed (pairing ID 0).
D (2079079) nvs: nvs_open_from_partition hap.A0 1
D (2079079) nvs: nvs_get_str_or_blob 00
D (2079119) nvs: nvs_close 167
D (2079119) nvs: nvs_open_from_partition hap.A0 1
D (2079119) nvs: nvs_get_str_or_blob 00
D (2079189) nvs: nvs_close 168
D (2079189) nvs: nvs_open_from_partition hap.A0 1
D (2079189) nvs: nvs_get_str_or_blob 00
D (2079249) nvs: nvs_close 169
D (2079259) nvs: nvs_open_from_partition hap.A0 1
D (2079259) nvs: nvs_get_str_or_blob 01
D (2079319) nvs: nvs_close 170
1970-01-01'T'00:34:38'Z'	Debug	[com.apple.mfi.HomeKit.Core:IPAccessoryServer] session:0x3ffb99d8:<
    0000  48545450 2f312e31 20323030 204f4b0d 0a436f6e 74656e74 2d547970 653a2061    HTTP/1.1 200 OK..Content-Type: a
    0020  70706c69 63617469 6f6e2f70 61697269 6e672b74 6c76380d 0a436f6e 74656e74    pplication/pairing+tlv8..Content
    0040  2d4c656e 6774683a 20330d0a 0d0a0601 04                                     -Length: 3.......
1970-01-01'T'00:34:38'Z'	Debug	[com.apple.mfi.HomeKit.Core:IPAccessoryServer] session:0x3ffb99d8:input closed
1970-01-01'T'00:34:38'Z'	Debug	[com.apple.mfi.HomeKit.Core:IPAccessoryServer] session:0x3ffb99d8:closing
1970-01-01'T'00:34:38'Z'	Debug	[com.apple.mfi.HomeKit.Core:IPAccessoryServer] session:0x3ffb99d8:closing security context
1970-01-01'T'00:34:38'Z'	Debug	[com.apple.mfi.HomeKit.Core:IPAccessoryServer] Closing HAP session.
1970-01-01'T'00:34:38'Z'	Debug	[com.apple.mfi.HomeKit.Core:Session] HAPSessionRelease
1970-01-01'T'00:34:38'Z'	Debug	[com.apple.mfi.HomeKit.Core:Session] HAPSessionInvalidate
1970-01-01'T'00:34:38'Z'	Debug	[com.apple.mfi.HomeKit.Core:IPAccessoryServer] session:0x3ffb99d8:closing TCP stream
1970-01-01'T'00:34:38'Z'	Debug	[com.apple.mfi.HomeKit.Platform:TCPStreamManager] close(50);
1970-01-01'T'00:34:38'Z'	Debug	[com.apple.mfi.HomeKit.Core:IPAccessoryServer] session:0x3ffb99d8:closed
1970-01-01'T'00:34:38'Z'	Debug	[com.apple.mfi.HomeKit.Core:IPAccessoryServer] session:0x3ffb99d8:releasing session
D (2079449) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3ffcb0dc
1970-01-01'T'00:34:39'Z'	Debug	[com.apple.mfi.HomeKit.Platform:TCPStreamManager] accept(49, NULL, NULL);
1970-01-01'T'00:34:39'Z'	Debug	[com.apple.mfi.HomeKit.Platform:TCPStreamManager] setsockopt(50, 6, 1, <buffer>);
    0000  01000000                                                                   ....
1970-01-01'T'00:34:39'Z'	Debug	[com.apple.mfi.HomeKit.Core:IPAccessoryServer] Opening HAP session.
1970-01-01'T'00:34:39'Z'	Debug	[com.apple.mfi.HomeKit.Core:Session] HAPSessionCreate
1970-01-01'T'00:34:39'Z'	Debug	[com.apple.mfi.HomeKit.Core:IPAccessoryServer] session:0x3ffb99d8:accepted
1970-01-01'T'00:34:39'Z'	Debug	[com.apple.mfi.HomeKit.Core:IPAccessoryServer] session:0x3ffb99d8:>
    0000  504f5354 202f7061 69722d76 65726966 79204854 54502f31 2e310d0a 486f7374    POST /pair-verify HTTP/1.1..Host
    0020  3a204163 6d655c30 33324c69 6768745c 30333242 756c622e 5f686170 2e5f7463    : Acme\032Light\032Bulb._hap._tc
    0040  702e6c6f 63616c0d 0a436f6e 74656e74 2d4c656e 6774683a 2033370d 0a436f6e    p.local..Content-Length: 37..Con
    0060  74656e74 2d547970 653a2061 70706c69 63617469 6f6e2f70 61697269 6e672b74    tent-Type: application/pairing+t
    0080  6c76380d 0a0d0a06 01010320 33409853 f12e0166 32c0d695 2094aed7 8532bcbc    lv8........ 3@.S...f2... ....2..
    00a0  1b43c561 be00d130 1fb00f74                                                 .C.a...0...t
1970-01-01'T'00:34:39'Z'	Debug	[com.apple.mfi.HomeKit.Core:PairingPairVerify] Pair Verify M1: Verify Start Request.
1970-01-01'T'00:34:39'Z'	Debug	[com.apple.mfi.HomeKit.Core:PairingPairVerify] Pair Verify M1: kTLVType_Method = 2.
1970-01-01'T'00:34:39'Z'	Debug	[com.apple.mfi.HomeKit.Core:PairingPairVerify] Pair Verify M1: Controller_cv_PK.
    0000  33409853 f12e0166 32c0d695 2094aed7 8532bcbc 1b43c561 be00d130 1fb00f74    3@.S...f2... ....2...C.a...0...t
1970-01-01'T'00:34:39'Z'	Debug	[com.apple.mfi.HomeKit.Core:PairingPairVerify] Pair Verify M2: Verify Start Response.
1970-01-01'T'00:34:39'Z'	Debug	[com.apple.mfi.HomeKit.Core:PairingPairVerify] <private> Pair Verify M2: cv_SK.
1970-01-01'T'00:34:39'Z'	Debug	[com.apple.mfi.HomeKit.Core:PairingPairVerify] Pair Verify M2: cv_PK.
    0000  6766dbe0 e376eef1 b2e052c7 734eb569 b0038f2b ab5a2a0a 5836eb5b 06d73421    gf...v....R.sN.i...+.Z*.X6.[..4!
1970-01-01'T'00:34:40'Z'	Debug	[com.apple.mfi.HomeKit.Core:PairingPairVerify] <private> Pair Verify M2: cv_KEY.
D (2081019) nvs: nvs_open_from_partition hap.90 1
D (2081019) nvs: nvs_get_str_or_blob 00
D (2081059) nvs: nvs_close 171
[... cut ...]
I didn't stated before, but there was a strange error during compilation. If it is related:

```
[1017/1034] Building C object esp-idf/port/CMakeFiles/__idf_port.dir/src/HAPPlatformLog.c.obj
/Users/Logic/esp32_test/esp-apple-homekit-adk/port/src/HAPPlatformLog.c: In function 'HAPPlatformLogPOSIXError':
/Users/Logic/esp32_test/esp-apple-homekit-adk/port/src/HAPPlatformLog.c:60:13: warning: initialization of 'int' from 'char *' makes integer from pointer without a cast [-Wint-conversion]
     int e = strerror_r(errorNumber, errorString, sizeof errorString);
             ^~~~~~~~~~
```

skanico, 2020-09-06:
#https://github.com/espressif/esp-apple-homekit-adk/pull/13/commits/3b10ee65e26f0dba635c06947cbb30a437f7f1b2 seems to solve the problem. No more crashes for me.