espressif/esp-apple-homekit-adk #6: Question? - where should I put ESP32 code to turn GPIO on/off

Vweston, 2020-06-29:
I’m unsure where I should put actual esp-idf code to say define GPIO pin for LED & actually set it on or off. Say for example I want to use the ESP32 built in led GPIO 2.would it go in the App.c or in app_main. Just a little help would go along way for new devs. 

The example works great from HomeKit perspective. Just need one more step.

Thanks.

shahpiyushv, 2020-06-30:
@Vweston It should go in App.c as it has the accessory specific logic. Specifically, AppInitialize(), AppDeinitialize(), IdentifyAccessory() and HandleLightBulbOnWrite() are the functions in App.c that you will have to modify.

Vweston, 2020-07-01:
Perfect, works great. Thank you

closed 2020-07-01 by Vweston

AramVartanyan, 2020-07-20:
@shahpiyushv What AppDeinitialize() function stands for?
In what circumstances it should be called?

shahpiyushv, 2020-07-21:
@AramVartanyan AppDeinitialize() is just a placeholder function wherein you can stop some modules or free some memory that was allocated in AppInitialize(). Most applications would not need it. Some use cases may require it to temporarily stop HomeKit to perform some other task, like say an OTA firmware upgrade, if the memory available is less.

AramVartanyan, 2020-07-21:
Thank you, @shahpiyushv. It is good know.

AleshR, 2020-12-26:
@shahpiyushv please, in your example "Lightbulb" a cannot find any GPIO definiton, maybe i'm blind... but i really need to change GPIO.