espressif/esp-apple-homekit-adk #7: Why the project needs to config the ssid and password , then can connect the apple homekit ?

xuhongv, 2020-07-10:
- Why the project needs to config the ssid and password , then can connect the apple homekit ?
- I understand the HomeKit Device  don't config the ssid the password in the code , and my HomeKit Device light don't need the step ?

shahpiyushv, 2020-07-10:
@xuhongv , HomeKit products use Apple's proprietary Wi-Fi Accessory Configuration (WAC) protocol for connecting to the Wi-Fi. However, that part of the specs and code is available only to MFi licensees. We have our [own SDKs](https://www.espressif.com/en/products/sdks/esp-homekit-sdk) which have the full HomeKit features, including WAC, but we are allowed to share them only with MFi licensees. You can. however, integrate our ESP IDF based Wi-Fi provisioning by referring [this example](https://github.com/espressif/esp-idf/tree/master/examples/provisioning/wifi_prov_mgr). The documentation for that is available [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/provisioning/wifi_provisioning.html) and the Phone apps are also [openly available](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/provisioning/wifi_provisioning.html#provisioning-tools).

xuhongv, 2020-07-13:
How can I get the Business Code ?

shahpiyushv, 2020-07-13:
@xuhongv , you will have to apply for the [Apple MFi program](https://developer.apple.com/programs/mfi/) and once you get it done, you can apply on [Espressif website](https://www.espressif.com/en/contact-us/sales-questions) to get the access. You will have to quote your MFi account number there.

xuhongv, 2020-07-13:
thanks @shahpiyushv

closed 2020-07-13 by xuhongv