espressif/esp-apple-homekit-adk #8: FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/app_wifi.c.obj

vicfrancco, 2020-07-10:
Hi!
During idf.py build, I'm receiving the following error.

Does anyone know how to fix it?

`[836/921] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/app_wifi.c.obj
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/app_wifi.c.obj
ccache C:\Users\victo\.espressif\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -Iconfig -I../main -IC:/Users/victo/Desktop/esp-idf/components/newlib/platform_include -IC:/Users/victo/Desktop/esp-idf/components/freertos/include -IC:/Users/victo/Desktop/esp-idf/components/heap/include -IC:/Users/victo/Desktop/esp-idf/components/log/include -IC:/Users/victo/Desktop/esp-idf/components/soc/esp32/include -IC:/Users/victo/Desktop/esp-idf/components/soc/include -IC:/Users/victo/Desktop/esp-idf/components/esp_rom/include -IC:/Users/victo/Desktop/esp-idf/components/esp_common/include -IC:/Users/victo/Desktop/esp-idf/components/xtensa/include -IC:/Users/victo/Desktop/esp-idf/components/xtensa/esp32/include -IC:/Users/victo/Desktop/esp-idf/components/esp32/include -IC:/Users/victo/Desktop/esp-idf/components/driver/include -IC:/Users/victo/Desktop/esp-idf/components/esp_ringbuf/include -IC:/Users/victo/Desktop/esp-idf/components/esp_event/include -IC:/Users/victo/Desktop/esp-idf/components/tcpip_adapter/include -IC:/Users/victo/Desktop/esp-idf/components/lwip/include/apps -IC:/Users/victo/Desktop/esp-idf/components/lwip/include/apps/sntp -IC:/Users/victo/Desktop/esp-idf/components/lwip/lwip/src/include -IC:/Users/victo/Desktop/esp-idf/components/lwip/port/esp32/include -IC:/Users/victo/Desktop/esp-idf/components/lwip/port/esp32/include/arch -IC:/Users/victo/Desktop/esp-idf/components/vfs/include -IC:/Users/victo/Desktop/esp-idf/components/esp_wifi/include -IC:/Users/victo/Desktop/esp-idf/components/esp_wifi/esp32/include -IC:/Users/victo/Desktop/esp-idf/components/esp_eth/include -IC:/Users/victo/Desktop/esp-idf/components/efuse/include -IC:/Users/victo/Desktop/esp-idf/components/efuse/esp32/include -IC:/Users/victo/Desktop/esp-idf/components/app_trace/include -IC:/Users/victo/Desktop/esp-idf/components/mbedtls/port/include -IC:/Users/victo/Desktop/esp-idf/components/mbedtls/mbedtls/include -IC:/Users/victo/Desktop/esp-idf/components/wpa_supplicant/include -IC:/Users/victo/Desktop/esp-idf/components/wpa_supplicant/port/include -IC:/Users/victo/Desktop/esp-idf/components/wpa_supplicant/include/esp_supplicant -IC:/Users/victo/Desktop/esp-idf/components/bootloader_support/include -IC:/Users/victo/Desktop/esp-idf/components/app_update/include -IC:/Users/victo/Desktop/esp-idf/components/spi_flash/include -IC:/Users/victo/Desktop/esp-idf/components/nvs_flash/include -IC:/Users/victo/Desktop/esp-idf/components/pthread/include -IC:/Users/victo/Desktop/esp-idf/components/espcoredump/include -IC:/Users/victo/Desktop/esp-idf/components/asio/asio/asio/include -IC:/Users/victo/Desktop/esp-idf/components/asio/port/include -IC:/Users/victo/Desktop/esp-idf/components/coap/port/include -IC:/Users/victo/Desktop/esp-idf/components/coap/port/include/coap -IC:/Users/victo/Desktop/esp-idf/components/coap/libcoap/include -IC:/Users/victo/Desktop/esp-idf/components/coap/libcoap/include/coap2 -IC:/Users/victo/Desktop/esp-idf/components/console -IC:/Users/victo/Desktop/esp-idf/components/nghttp/port/include -IC:/Users/victo/Desktop/esp-idf/components/nghttp/nghttp2/lib/includes -IC:/Users/victo/Desktop/esp-idf/components/esp-tls -IC:/Users/victo/Desktop/esp-idf/components/esp_adc_cal/include -IC:/Users/victo/Desktop/esp-idf/components/esp_gdbstub/include -IC:/Users/victo/Desktop/esp-idf/components/tcp_transport/include -IC:/Users/victo/Desktop/esp-idf/components/esp_http_client/include -IC:/Users/victo/Desktop/esp-idf/components/esp_http_server/include -IC:/Users/victo/Desktop/esp-idf/components/esp_https_ota/include -IC:/Users/victo/Desktop/esp-idf/components/protobuf-c/protobuf-c -IC:/Users/victo/Desktop/esp-idf/components/protocomm/include/common -IC:/Users/victo/Desktop/esp-idf/components/protocomm/include/security -IC:/Users/victo/Desktop/esp-idf/components/protocomm/include/transports -IC:/Users/victo/Desktop/esp-idf/components/mdns/include -IC:/Users/victo/Desktop/esp-idf/components/esp_local_ctrl/include -IC:/Users/victo/Desktop/esp-idf/components/esp_websocket_client/include -IC:/Users/victo/Desktop/esp-idf/components/expat/expat/expat/lib -IC:/Users/victo/Desktop/esp-idf/components/expat/port/include -IC:/Users/victo/Desktop/esp-idf/components/wear_levelling/include -IC:/Users/victo/Desktop/esp-idf/components/sdmmc/include -IC:/Users/victo/Desktop/esp-idf/components/fatfs/diskio -IC:/Users/victo/Desktop/esp-idf/components/fatfs/vfs -IC:/Users/victo/Desktop/esp-idf/components/fatfs/src -IC:/Users/victo/Desktop/esp-idf/components/freemodbus/common/include -IC:/Users/victo/Desktop/esp-idf/components/idf_test/include -IC:/Users/victo/Desktop/esp-idf/components/jsmn/include -IC:/Users/victo/Desktop/esp-idf/components/json/cJSON -IC:/Users/victo/Desktop/esp-idf/components/libsodium/libsodium/src/libsodium/include -IC:/Users/victo/Desktop/esp-idf/components/libsodium/port_include -IC:/Users/victo/Desktop/esp-idf/components/mqtt/esp-mqtt/include -IC:/Users/victo/Desktop/esp-idf/components/openssl/include -IC:/Users/victo/Desktop/esp-idf/components/spiffs/include -IC:/Users/victo/Desktop/esp-idf/components/ulp/include -IC:/Users/victo/Desktop/esp-idf/components/unity/include -IC:/Users/victo/Desktop/esp-idf/components/unity/unity/src -IC:/Users/victo/Desktop/esp-idf/components/wifi_provisioning/include -IC:/Users/victo/esp/esp-apple-homekit-adk/port/include -IC:/Users/victo/esp/esp-apple-homekit-adk/homekit_adk/HAP -IC:/Users/victo/esp/esp-apple-homekit-adk/homekit_adk/PAL -IC:/Users/victo/esp/esp-apple-homekit-adk/homekit_adk/External/Base64 -IC:/Users/victo/esp/esp-apple-homekit-adk/homekit_adk/External/JSON -IC:/Users/victo/esp/esp-apple-homekit-adk/homekit_adk/External/HTTP -mlongcalls -Wno-frame-address   -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.0.1-dirty\" -DGCC_NOT_5_2_0 -DESP_PLATFORM -MD -MT esp-idf/main/CMakeFiles/__idf_main.dir/app_wifi.c.obj -MF esp-idf\main\CMakeFiles\__idf_main.dir\app_wifi.c.obj.d -o esp-idf/main/CMakeFiles/__idf_main.dir/app_wifi.c.obj   -c ../main/app_wifi.c
In file included from C:/Users/victo/Desktop/esp-idf/components/esp_common/include/esp_timer.h:44,
                 from C:/Users/victo/Desktop/esp-idf/components/freertos/include/freertos/portmacro.h:84,
                 from C:/Users/victo/Desktop/esp-idf/components/freertos/include/freertos/portable.h:96,
                 from C:/Users/victo/Desktop/esp-idf/components/freertos/include/freertos/FreeRTOS.h:105,
                 from ../main/app_wifi.c:10:
../main/app_wifi.c: In function 'app_wifi_init':
../main/app_wifi.c:71:21: error: implicit declaration of function 'esp_netif_init'; did you mean 'esp_wifi_init'? [-Werror=implicit-function-declaration]
     ESP_ERROR_CHECK(esp_netif_init());
                     ^~~~~~~~~~~~~~
C:/Users/victo/Desktop/esp-idf/components/esp_common/include/esp_err.h:118:31: note: in definition of macro 'ESP_ERROR_CHECK'
         esp_err_t __err_rc = (x);                                       \
                               ^
../main/app_wifi.c:73:5: error: implicit declaration of function 'esp_netif_create_default_wifi_sta' [-Werror=implicit-function-declaration]
     esp_netif_create_default_wifi_sta();
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../main/app_wifi.c: In function 'app_wifi_connect':
../main/app_wifi.c:82:5: error: unknown type name 'esp_event_handler_instance_t'; did you mean 'esp_event_handler_t'?
     esp_event_handler_instance_t instance_any_id;
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
     esp_event_handler_t
../main/app_wifi.c:83:5: error: unknown type name 'esp_event_handler_instance_t'; did you mean 'esp_event_handler_t'?
     esp_event_handler_instance_t instance_got_ip;
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
     esp_event_handler_t
In file included from C:/Users/victo/Desktop/esp-idf/components/esp_common/include/esp_timer.h:44,
[... cut ...] /Desktop/esp-idf/components/bootloader/subproject/components/micro-ecc C:/Users/victo/Desktop/esp-idf/components/partition_table C:/Users/victo/Desktop/esp-idf/components/soc C:/Users/victo/Desktop/esp-idf/components/spi_flash C:/Users/victo/Desktop/esp-idf/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/victo/esp/esp-apple-homekit-adk/examples/Lightbulb/build/bootloader
ninja: build stopped: subcommand failed.
ninja failed with exit code 1`

shahpiyushv, 2020-07-11:
@vicfrancco , are you using some older esp idf? esp-apple-homekit-sdk is supposed to work with release/4.2 or master branches.

vicfrancco, 2020-07-12:
> 
> 
> @vicfrancco , are you using some older esp idf? esp-apple-homekit-sdk is supposed to work with release/4.2 or master branches.

Hi shahpiyushv!

I think it's the master branch, because I've installed ESP-IDF Installer Tools for Windows.

When I open ESP-IDF Command Prompt, it shows the following info:

`Using Python in C:\Users\victo\AppData\Local\Programs\Python\Python38-32\
Python 3.8.3
Using Git in C:\Program Files\Git\cmd\
git version 2.21.0.windows.1
Setting IDF_PATH: C:\Users\victo\Desktop\esp-idf

Adding ESP-IDF tools to PATH...
    C:\Users\victo\.espressif\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\bin
    C:\Users\victo\.espressif\tools\esp32ulp-elf\2.28.51.20170517\esp32ulp-elf-binutils\bin
    C:\Users\victo\.espressif\tools\cmake\3.13.4\bin
    C:\Users\victo\.espressif\tools\openocd-esp32\v0.10.0-esp32-20190313\openocd-esp32\bin
    C:\Users\victo\.espressif\tools\mconf\v4.6.0.0-idf-20190628\
    C:\Users\victo\.espressif\tools\ninja\1.9.0\
    C:\Users\victo\.espressif\tools\idf-exe\1.0.1\
    C:\Users\victo\.espressif\tools\ccache\3.7\
    C:\Users\victo\.espressif\python_env\idf4.0_py3.8_env\Scripts
    C:\Users\victo\Desktop\esp-idf\tools

Checking if Python packages are up to date...
Python requirements from C:\Users\victo\Desktop\esp-idf\requirements.txt are satisfied.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build`

shahpiyushv, 2020-07-13:
@vicfrancco , this part of your log indicates that you may be using idf 4.0

```
    C:\Users\victo\.espressif\python_env\idf4.0_py3.8_env\Scripts
```

Can you re-run the installer and get the master branch of esp-idf, instead of release/v4.0?

PIYUSHGIRI2021, 2021-05-31:
Hii,@shahpiyushv
i am also getting same error and other error also ,please let me know how can i resolve it,thanks

 C:\Users\Frank\Desktop\esp-idf\examples\piyush\EVQPoint> idf.py build
Executing action: all (aliases: build)
Running ninja in directory c:\users\frank\desktop\esp-idf\examples\piyush\evqpoint\build
Executing "ninja all"...
[1/12] Performing build step for 'bootloader'
ninja: no work to do.
[2/10] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/controller_main.c.obj
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/controller_main.c.obj
ccache C:\Users\Frank\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -Iconfig -I../../../../components/newlib/platform_include -I../../../../components/freertos/include -I../../../../components/freertos/xtensa/include -I../../../../components/heap/include -I../../../../components/log/include -I../../../../components/lwip/include/apps -I../../../../components/lwip/include/apps/sntp -I../../../../components/lwip/lwip/src/include -I../../../../components/lwip/port/esp32/include -I../../../../components/lwip/port/esp32/include/arch -I../../../../components/soc/src/esp32/. -I../../../../components/soc/src/esp32/include -I../../../../components/soc/include -I../../../../components/esp_rom/include -I../../../../components/esp_common/include -I../../../../components/esp_system/include -I../../../../components/xtensa/include -I../../../../components/xtensa/esp32/include -I../../../../components/esp32/include -I../../../../components/driver/include -I../../../../components/driver/esp32/include -I../../../../components/esp_ringbuf/include -I../../../../components/efuse/include -I../../../../components/efuse/esp32/include -I../../../../components/espcoredump/include -I../../../../components/esp_timer/include -I../../../../components/esp_ipc/include -I../../../../components/soc/soc/esp32/include -I../../../../components/soc/soc/esp32/../include -I../../../../components/soc/soc/esp32/private_include -I../../../../components/vfs/include -I../../../../components/esp_wifi/include -I../../../../components/esp_wifi/esp32/include -I../../../../components/esp_event/include -I../../../../components/esp_netif/include -I../../../../components/esp_eth/include -I../../../../components/tcpip_adapter/include -I../../../../components/app_trace/include -I../../../../components/mbedtls/port/include -I../../../../components/mbedtls/mbedtls/include -I../../../../components/mbedtls/esp_crt_bundle/include -I../../../../components/bootloader_support/include -I../../../../components/app_update/include -I../../../../components/spi_flash/include -I../../../../components/wpa_supplicant/include -I../../../../components/wpa_supplicant/port/include -I../../../../components/wpa_supplicant/include/esp_supplicant -I../../../../components/nvs_flash/include -I../../../../components/pthread/include -I../../../../components/perfmon/include -I../../../../components/asio/asio/asio/include -I../../../../components/asio/port/include -I../../../../components/cbor/port/include -I../../../../components/coap/port/include -I../../../../components/coap/port/include/coap -I../../../../components/coap/libcoap/include -I../../../../components/coap/libcoap/include/coap2 -I../../../../components/console -I../../../../components/nghttp/port/include -I../../../../components/nghttp/nghttp2/lib/includes -I../../../../components/esp-tls -I../../../../components/esp_adc_cal/include -I../../../../components/esp_gdbstub/include -I../../../../components/esp_hid/include -I../../../../components/tcp_transport/include -I../../../../components/esp_http_client/include -I../../../../components/esp_http_server/include -I../../../../components/esp_https_ota/include -I../../../../components/protobuf-c/protobuf-c -I../../../../components/protocomm/include/common -I../../../../components/protocomm/include/security -I../../../../components/protocomm/include/transports -I../../../../components/mdns/include -I../../../../components/esp_local_ctrl/include -I../../../../components/sdmmc/include -I../../../../components/esp_serial_slave_link/include -I../../../../components/esp_websocket_client/include -I../../../../components/expat/expat/expat/lib -I../../../../components/expat/port/include -I../../../../components/wear_levelling/include -I../../../../components/fatfs/diskio -I../../../../components/fatfs/vfs -I../../../../components/fatfs/src -I../../../../components/freemodbus/common/include -I../../../../components/idf_test/include -I../../../../components/idf_test/include/esp32 -I../../../../components/jsmn/include -I../../../../components/json/cJSON -I../../../../components/libsodium/libsodium/src/libsodium/include -I../../../../components/libsodium/port_include -I../../../../components/mqtt/esp-mqtt/include -I../../../../components/openssl/include -I../../../../components/spiffs/include -I../../../../components/ulp/include -I../../../../components/unity/include -I../../../../components/unity/unity/src -I../../../../components/wifi_provisioning/include -I../components/modbus/libmodbus-3.1.4/src -mlongcalls -Wno-frame-address   -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggd [... cut ...] -MF esp-idf\modbus\CMakeFiles\__idf_modbus.dir\libmodbus-3.1.4\src\modbus.c.obj.d -o esp-idf/modbus/CMakeFiles/__idf_modbus.dir/libmodbus-3.1.4/src/modbus.c.obj   -c ../components/modbus/libmodbus-3.1.4/src/modbus.c
../components/modbus/libmodbus-3.1.4/src/modbus.c:27:10: fatal error: modbus_slave_service.h: No such file or directory
 #include "modbus_slave_service.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

hassanaansari, 2024-09-06:
I am also facing the same issue 
[963/970] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/wifi.c.obj

It stops here and much more down the line after idf.py build