espressif/esp-apple-homekit-adk #9: QR code生成

liuqiangqrs, 2020-07-17:
homekit的二维码怎么生成呢？
https://www.studiopieters.nl/esp-homekit-sdk-qr-code/
https://github.com/maximkulkin/esp-homekit

这两个链接我参考上边的方法做出来homekit可以识别二维码，但是不能用，不能添加设备。

[image]

shahpiyushv, 2020-07-17:
@liuqiangqrs have you used the correct setup id and accessory category while generating the QR code? Similar QR can also be generated by following the steps [here](https://github.com/espressif/esp-apple-homekit-adk#change-the-setup-code). The information already encoded in accessory_setup.bin has been generated using this:

```
$ ./provision_raspi.sh --category 5 --setup-code 111-22-333 --setup-id ES32 Lightbulb
```

liuqiangqrs, 2020-07-20:
如果固件中的setup_id 和生成二维码的setup_id不一样会导致homekit不能成功给esp32配网吗？我看资料说setup_id是可以是一个随机值。我参考的链接的方法有问题吗？

shahpiyushv, 2021-01-07:
@liuqiangqrs , sorry this wasn't answered earlier. setup_id from QR code needs to match the one programmed in accessory, even though it is random. Were you able to move past this issue?