espressif/esp-apple-homekit-adk #10: <netinet/tcp.h> No such file or directory

Asteliks, 2020-07-18:
When trying to compile the example sketch I het two errors:

1. 

`[9/98] ccache C:\ESP\.espressif\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -DHAP_LOG_LEVEL=1 -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -Iconfig -IC:/ESP/esp-idf/components/port/include -IC:/ESP/esp-idf/components/homekit_adk/HAP -IC:/ESP/esp-idf/components/homekit_adk/PAL -IC:/ESP/esp-idf/components/homekit_adk/External/Base64 -IC:/ESP/esp-idf/components/homekit_adk/External/JSON -IC:/ESP/esp-idf/components/homekit_adk/External/HTTP -IC:/ESP/esp-idf/components/newlib/platform_include -IC:/ESP/esp-idf/components/freertos/include -IC:/ESP/esp-idf/components/heap/include -IC:/ESP/esp-idf/components/log/include -IC:/ESP/esp-idf/components/soc/esp32/include -IC:/ESP/esp-idf/components/soc/include -IC:/ESP/esp-idf/components/esp_rom/include -IC:/ESP/esp-idf/components/esp_common/include -IC:/ESP/esp-idf/components/xtensa/include -IC:/ESP/esp-idf/components/xtensa/esp32/include -IC:/ESP/esp-idf/components/esp32/include -IC:/ESP/esp-idf/components/driver/include -IC:/ESP/esp-idf/components/esp_ringbuf/include -IC:/ESP/esp-idf/components/esp_event/include -IC:/ESP/esp-idf/components/tcpip_adapter/include -IC:/ESP/esp-idf/components/lwip/include/apps -IC:/ESP/esp-idf/components/lwip/include/apps/sntp -IC:/ESP/esp-idf/components/lwip/lwip/src/include -IC:/ESP/esp-idf/components/lwip/port/esp32/include -IC:/ESP/esp-idf/components/lwip/port/esp32/include/arch -IC:/ESP/esp-idf/components/vfs/include -IC:/ESP/esp-idf/components/esp_wifi/include -IC:/ESP/esp-idf/components/esp_wifi/esp32/include -IC:/ESP/esp-idf/components/esp_eth/include -IC:/ESP/esp-idf/components/efuse/include -IC:/ESP/esp-idf/components/efuse/esp32/include -IC:/ESP/esp-idf/components/app_trace/include -IC:/ESP/esp-idf/components/nvs_flash/include -IC:/ESP/esp-idf/components/spi_flash/include -IC:/ESP/esp-idf/components/mbedtls/port/include -IC:/ESP/esp-idf/components/mbedtls/mbedtls/include -IC:/ESP/esp-idf/components/mdns/include -IC:/ESP/esp-idf/components/console -mlongcalls -Wno-frame-address   -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.0.1-dirty\" -DGCC_NOT_5_2_0 -DESP_PLATFORM -MD -MT esp-idf/port/CMakeFiles/__idf_port.dir/src/HAPPlatformTCPStreamManager.c.obj -MF esp-idf\port\CMakeFiles\__idf_port.dir\src\HAPPlatformTCPStreamManager.c.obj.d -o esp-idf/port/CMakeFiles/__idf_port.dir/src/HAPPlatformTCPStreamManager.c.obj   -c C:/ESP/esp-idf/components/port/src/HAPPlatformTCPStreamManager.c
FAILED: esp-idf/port/CMakeFiles/__idf_port.dir/src/HAPPlatformTCPStreamManager.c.obj 
ccache C:\ESP\.espressif\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -DHAP_LOG_LEVEL=1 -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -Iconfig -IC:/ESP/esp-idf/components/port/include -IC:/ESP/esp-idf/components/homekit_adk/HAP -IC:/ESP/esp-idf/components/homekit_adk/PAL -IC:/ESP/esp-idf/components/homekit_adk/External/Base64 -IC:/ESP/esp-idf/components/homekit_adk/External/JSON -IC:/ESP/esp-idf/components/homekit_adk/External/HTTP -IC:/ESP/esp-idf/components/newlib/platform_include -IC:/ESP/esp-idf/components/freertos/include -IC:/ESP/esp-idf/components/heap/include -IC:/ESP/esp-idf/components/log/include -IC:/ESP/esp-idf/components/soc/esp32/include -IC:/ESP/esp-idf/components/soc/include -IC:/ESP/esp-idf/components/esp_rom/include -IC:/ESP/esp-idf/components/esp_common/include -IC:/ESP/esp-idf/components/xtensa/include -IC:/ESP/esp-idf/components/xtensa/esp32/include -IC:/ESP/esp-idf/components/esp32/include -IC:/ESP/esp-idf/components/driver/include -IC:/ESP/esp-idf/components/esp_ringbuf/include -IC:/ESP/esp-idf/components/esp_event/include -IC:/ESP/esp-idf/components/tcpip_adapter/include -IC:/ESP/esp-idf/components/lwip/include/apps -IC:/ESP/esp-idf/components/lwip/include/apps/sntp -IC:/ESP/esp-idf/components/lwip/lwip/src/include -IC:/ESP/esp-idf/components/lwip/port/esp32/include -IC:/ESP/esp-idf/components/lwip/port/esp32/include/arch -IC:/ESP/esp-idf/components/vfs/include -IC:/ESP/esp-idf/components/esp_wifi/include -IC:/ESP/esp-idf/components/esp_wifi/esp32/include -IC:/ESP/esp-idf/components/esp_eth/include -IC:/ESP/esp-idf/components/efuse/include -IC:/ESP/esp-idf/components/efuse/esp32/include -IC:/ESP/esp-idf/components/app_trace/include -IC:/ESP/esp-idf/components/nvs_flash/include -IC:/ESP/esp-idf/components/spi_flash/include -IC:/ESP/esp-idf/components/mbedtls/port/include -IC:/ESP/esp-idf/components/mbedtls/mbedtls/include -IC:/ESP/esp-idf/components/mdns/include -IC:/ESP/esp-idf/components/console -mlongcalls -Wno-frame-address   -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.0.1-dirty\" -DGCC_NOT_5_2_0 -DESP_PLATFORM -MD -MT esp-idf/port/CMakeFiles/__idf_port.dir/src/HAPPlatformTCPStreamManager.c.obj -MF esp-idf\port\CMakeFiles\__idf_port.dir\src\HAPPlatformTCPStreamManager.c.obj.d -o esp-idf/port/CMakeFiles/__idf_port.dir/src/HAPPlatformTCPStreamManager.c.obj   -c C:/ESP/esp-idf/components/port/src/HAPPlatformTCPStreamManager.c
C:/ESP/esp-idf/components/port/src/HAPPlatformTCPStreamManager.c:25:10: fatal error: netinet/tcp.h: No such file or directory
 #include <netinet/tcp.h>`

2. 

`FAILED: esp-idf/port/CMakeFiles/__idf_port.dir/__/homekit_adk/PAL/Crypto/MbedTLS/HAPMbedTLS.c.obj 
ccache C:\ESP\.espressif\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -DHAP_LOG_LEVEL=1 -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -Iconfig -IC:/ESP/esp-idf/components/port/include -IC:/ESP/esp-idf/components/homekit_adk/HAP -IC:/ESP/esp-idf/components/homekit_adk/PAL -IC:/ESP/esp-idf/components/homekit_adk/External/Base64 -IC:/ESP/esp-idf/components/homekit_adk/External/JSON -IC:/ESP/esp-idf/components/homekit_adk/External/HTTP -IC:/ESP/esp-idf/components/newlib/platform_include -IC:/ESP/esp-idf/components/freertos/include -IC:/ESP/esp-idf/components/heap/include -IC:/ESP/esp-idf/components/log/include -IC:/ESP/esp-idf/components/soc/esp32/include -IC:/ESP/esp-idf/components/soc/include -IC:/ESP/esp-idf/components/esp_rom/include -IC:/ESP/esp-idf/components/esp_common/include -IC:/ESP/esp-idf/components/xtensa/include -IC:/ESP/esp-idf/components/xtensa/esp32/include -IC:/ESP/esp-idf/components/esp32/include -IC:/ESP/esp-idf/components/driver/include -IC:/ESP/esp-idf/components/esp_ringbuf/include -IC:/ESP/esp-idf/components/esp_event/include -IC:/ESP/esp-idf/components/tcpip_adapter/include -IC:/ESP/esp-idf/components/lwip/include/apps -IC:/ESP/esp-idf/components/lwip/include/apps/sntp -IC:/ESP/esp-idf/components/lwip/lwip/src/include -IC:/ESP/esp-idf/components/lwip/port/esp32/include -IC:/ESP/esp-idf/components/lwip/port/esp32/include/arch -IC:/ESP/esp-idf/components/vfs/include -IC:/ESP/esp-idf/components/esp_wifi/include -IC:/ESP/esp-idf/components/esp_wifi/esp32/include -IC:/ESP/esp-idf/components/esp_eth/include -IC:/ESP/esp-idf/components/efuse/include -IC:/ESP/esp-idf/components/efuse/esp32/include -IC:/ESP/esp-idf/components/app_trace/include -IC:/ESP/esp-idf/components/nvs_flash/include -IC:/ESP/esp-idf/components/spi_flash/include -IC:/ESP/esp-idf/components/mbedtls/port/include -IC:/ESP/esp-idf/components/mbedtls/mbedtls/include -IC:/ESP/esp-idf/components/mdns/include -IC:/ESP/esp-idf/components/console -mlongcalls -Wno-frame-address   -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.0.1-dirty\" -DGCC_NOT_5_2_0 -DESP_PLATFORM -MD -MT esp-idf/port/CMakeFiles/__idf_port.dir/__/homekit_adk/PAL/Crypto/MbedTLS/HAPMbedTLS.c.obj -MF esp-idf\port\CMakeFiles\__idf_port.dir\__\homekit_adk\PAL\Crypto\MbedTLS\HAPMbedTLS.c.obj.d -o esp-idf/p [... cut ...] pto/MbedTLS/HAPMbedTLS.c
C:/ESP/esp-idf/components/homekit_adk/PAL/Crypto/MbedTLS/HAPMbedTLS.c: In function 'HAP_X25519_scalarmult':
C:/ESP/esp-idf/components/homekit_adk/PAL/Crypto/MbedTLS/HAPMbedTLS.c:119:19: error: implicit declaration of function 'mbedtls_ecp_read_key'; did you mean 'mbedtls_ecp_gen_key'? [-Werror=implicit-function-declaration]
         int ret = mbedtls_ecp_read_key(MBEDTLS_ECP_DP_CURVE25519, &our_key, n, X25519_SCALAR_BYTES);`

Unfortunatly I have been unable to solve them

shahpiyushv, 2020-07-18:
@Asteliks , have you copied the repo into esp-idf components? I haven't tried compiling that way. Can you follow the instructions from the [README](https://github.com/espressif/esp-apple-homekit-adk#compile-and-flash) instead and try?

For the second issue, please check the [mBedTLS setup](https://github.com/espressif/esp-apple-homekit-adk#set-up-mbedtls) instructions in the README.

Asteliks, 2020-07-18:
I have copied homekit_adk and port into esp-idf components as I'm using eclipse.
Well I tried following the instructions but I'm failing to complete the Compile and Flash step.

for the second issue for some reason mBedTLS went back to 2.16.5. Well that is interesting, Also for some reason following the steps for mBedTLS via Git runs me into other issues. I've solved that downloading the zip. To be exact this:
`axels@MSI-Axel MINGW64 /c/ESP/esp-idf/components/mbedtls/mbedtls ((9ef92c551...))
$ git pull
You are not currently on a branch.
Please specify which branch you want to merge with.
See git-pull(1) for details.

    git pull <remote> <branch>

axels@MSI-Axel MINGW64 /c/ESP/esp-idf/components/mbedtls/mbedtls ((9ef92c551...))
$ git checkout -b mbedtls-2.16.6-adk origin/mbedtls-2.16.6-adk
error: Your local changes to the following files would be overwritten by checkout:
        scripts/output_env.sh
        tests/scripts/all.sh
        tests/scripts/check-python-files.sh
        tests/ssl-opt.sh
Please commit your changes or stash them before you switch branches.
error: The following untracked working tree files would be overwritten by checkout:
        dco.txt
Please move or remove them before you switch branches.
Aborting
`

Asteliks, 2020-07-20:
It would seem that for some reason esp-idf 4.0.1 is missing tcp.h link: [https://github.com/espressif/esp-idf/tree/v4.0.1/components/lwip/port/esp32/include/netinet](url)

tcp.h is present in master link: [https://github.com/espressif/esp-idf/tree/master/components/lwip/port/esp32/include/netinet](url)

shahpiyushv, 2020-07-21:
@Asteliks For the mbedtls error, please follow instructions as mentioned. Use `git stash` to stash any local changes, or just delete those files before switching branches.

closed 2020-07-25 by Asteliks