espressif/esp-apple-homekit-adk #14: Where to generate your own example?

kistanod, 2020-08-18:
I am working on a way to generate custom homekit items, and I would like to ask how files for Lightbulb directories were created in the first place? What kind of tools were used! thank you very much

shahpiyushv, 2020-08-25:
The file for sample was taken directly from [Apple's HomeKit ADK Sample](https://github.com/apple/HomeKitADK/tree/5c19c9eb2bff41115ec8f1c763c71bd6a69545e5/Applications/Lightbulb).  I myself am not aware of any tool to generate it.

shahpiyushv, 2020-09-25:
@kistanovdev , we have also opened up our own implementation of HomeKit [here](https://github.com/espressif/esp-homekit-sdk). Have a look to check if that suits your requirement better.

shahpiyushv, 2020-10-11:
Closing this issue as it has been answered.

closed 2020-10-11 by shahpiyushv