espressif/esp-apple-homekit-adk #16: Not lighting up even though successful execution

weilie91, 2020-09-05:
Hi, I am new to this. I have followed the instructions thoroughly and managed to proceed without any issue, except that the light is not lighting up when I switch it on homekit. I tried finding the file to check the output(which I assumed it's the inbuilt LED) but to no avail. Please advise. Thanks

shahpiyushv, 2020-09-05:
@weilie91 , please check [this function in App.c](https://github.com/espressif/esp-apple-homekit-adk/blob/868fd5be7f9af4252877c2e7c9c5451ea7de8c19/examples/Lightbulb/main/App.c#L163) which is the callback. You can add your hardware logic there. ESP32-DevKit-C, which is our most common boar,d does not have in-built user controllable LED, and so the example too does not have any hardware control.

emanuelelaface, 2020-12-25:
Is there a simple GPIO blinking led example? I am familiar with Arduino (and C) but I have no idea how to start to control the GPIOs for this board from the App.c function.
As far as I understood the callbacks are `HandleLightBulbOnWrite` to change status and `HandleLightBulbOnRead` to get the status, but how can I switch on and off a pin? Could you please point me to a documentation to understand what I am suppose to write there?
Many thanks and sorry for the dumb question.

emanuelelaface, 2020-12-26:
Ok I find what to add, I report here because it can be useful for other beginners with ESP32 like me. Maybe you could add this (commented?) in the App.c as example of how to turn on the GPIO.

Everything goes in `App.c`.
First thing is to include the header for GPIOs, I also added a define for the name of the GPIO to switch:
```
#include "driver/gpio.h"
#define LED_GPIO 2
```
then you need to setup at the startup the GPIO as output. I am not sure where is the better place to put this code, but I setup it in the `AppAccessoryServerStart` function that runs before starting the app, maybe it can be done in a better place. At the end of the function I added:
```
gpio_reset_pin(LED_GPIO);
gpio_set_direction(LED_GPIO, GPIO_MODE_OUTPUT);
```
finally the logic to turn it on on off stays in the `HandleLightBulbOnWrite` and has to change the value of the led as function of the `value`:
```
if (value) {
    gpio_set_level(LED_GPIO, 1);
} else {
    gpio_set_level(LED_GPIO, 0);
}
```
That's all. This will turn on and off the GPIO pin with the number 2 on the board (on some board where the red led is connected to number 2 you will see that one blinking as well, mine is not connected to any GPIO).

AramVartanyan, 2020-12-26:
@emanuelelaface it is better to use this example to setup the output:
https://github.com/espressif/esp-idf/tree/master/examples/peripherals/gpio/generic_gpio

In this way you can avoid possible unwanted behavior of the I/O peripherals.

emanuelelaface, 2020-12-26:
Thanks, I see that it may be safer to initialize it in the proper way. Where would you put the init part in the homekit example? I sued the `AppAccessoryServerStart` but it doesn't sound as the correct place (it should be used for the server) but on the other hand there is no initializer there for the peripherals.

AramVartanyan, 2020-12-26:
For me the proper place would be in the InitializePlatform function inside app_main.c