espressif/esp-apple-homekit-adk #17: How to serve multiple accessories?

whymidnight, 2020-10-15:
In the Lightbulb example only covers one accessory.

I have a single ESP32 that controls 8 GPIO pins that I want to operate as independent accessories. 

Would this be possible?

skanico, 2020-10-15:
Yes it is possible. You can see examples [here](https://github.com/Asteliks/AppleHomeKitADK-ESP32)

shahpiyushv, 2020-10-15:
@skanico thanks for a pointer to the example. Meanwhile @whymidnight, Apple general expects these to be services on the same accessory as these are co-located. But since this may not be an end product, it's fine :)

Meanwhile, have you checked out our other SDK, the [esp-homekit-sdk](https://github.com/espressif/esp-homekit-sdk) ? It is our own implementation with a different set of APIs. We have multiple examples there, including bridges.

whymidnight, 2020-10-15:
The [esp-homekit-sdk](https://github.com/espressif/esp-homekit-sdk) is the way to go for this. Thank you @shahpiyushv.

closed 2020-10-15 by whymidnight