espressif/esp-apple-homekit-adk #19: LightDemo can't be discoverd by iphone

mykonica, 2020-10-30:
follow the guide to build, but some error happen, i don't known why.
**1970-01-01'T'00:00:00'Z'	Default	[com.apple.mfi.HomeKit.Platform:KeyValueStore] Error (4354). Key 20 not found in KeyStore
1970-01-01'T'00:00:00'Z'	Default	[com.apple.mfi.HomeKit.Platform:KeyValueStore] Error (4354). Key 21 not found in KeyStore**

--------------------------------------------------------------------------

**1970-01-01'T'00:00:00'Z'	Default	[com.apple.mfi.HomeKit.Platform:KeyValueStore] Error (4354). Key 20 not found in KeyStore
1970-01-01'T'00:00:00'Z'	Default	[com.apple.mfi.HomeKit.Platform:KeyValueStore] Error (4354). Key 21 not found in KeyStore**
I (1019) phy: phy_version: 4500, 0cd6843, Sep 17 2020, 15:37:07, 0, 0
I (1019) wifi:mode : sta (10:52:1c:68:98:08)
I (1019) wifi:enable tsf
I (1019) wifi station: wifi_init_sta finished.
I (1149) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (1149) wifi:state: init -> auth (b0)
I (1149) wifi:state: auth -> assoc (0)
I (1159) wifi:state: assoc -> run (10)
I (1159) wifi:connected with Tencent-GuestWiFi, aid = 62, channel 1, BW20, bssid = 00:2a:10:09:bb:b3
I (1159) wifi:security: Open Auth, phy: bgn, rssi: -67
I (1169) wifi:pm start, type: 1

W (1179) wifi:<ba-add>idx:0 (ifx:0, 00:2a:10:09:bb:b3), tid:0, ssn:3206, winSize:64
I (1299) wifi:AP's beacon interval = 104448 us, DTIM period = 1
I (1789) esp_netif_handlers: sta ip: 10.35.192.76, mask: 255.255.255.0, gw: 10.35.192.1
I (1789) wifi station: got ip:10.35.192.76
I (1789) wifi station: connected to ap SSID:Tencent-GuestWiFi password:
1970-01-01'T'00:00:01'Z'	Default	[com.apple.mfi.HomeKit.Core:AccessoryServer] Version information:
libhap: Unknown
  - Version: Internal (Oct 30 2020 20:02:24) - compatibility version 7
Using platform: ESP32
  - Version: v4.3-dev-1561-g357a27760-dirty (Oct 30 2020 20:02:32) - compatibility version 7
  - Available features:
    - Key-Value store
    - Accessory setup manager
    - Service discovery
**1970-01-01'T'00:00:01'Z'	Default	[com.apple.mfi.HomeKit.Platform:KeyValueStore] Error (4354). Key 00 not found in KeyStore**
1970-01-01'T'00:00:01'Z'	Default	[com.apple.mfi.HomeKit.Platform:Clock] Using 'gettimeofday'.

shahpiyushv, 2020-10-30:
@mykonica you may have missed the step `esptool.py -p $ESPPORT write_flash 0x340000 accessory_setup.bin` under [Compile and Flash](https://github.com/espressif/esp-apple-homekit-adk#compile-and-flash). Can you double check?

mykonica, 2020-10-30:
> @mykonica you may have missed the step `esptool.py -p $ESPPORT write_flash 0x340000 accessory_setup.bin` under [Compile and Flash](https://github.com/espressif/esp-apple-homekit-adk#compile-and-flash). Can you double check?

i do the "esptool.py -p $ESPPORT write_flash 0x340000 accessory_setup.bin", but problem is still exist. 
the accessory_setup.csv file has no key 20、21、00, is it right ?
[image]

mykonica, 2020-10-30:
i test it again，although the error exists, but i can discover the device with iphone Home App。@shahpiyushv

shahpiyushv, 2020-10-30:
@mykonica , in that case, you may ignore the errors. On boot up, the ADK tries to read some variables from persistent storage and depending on their presence/absence, sets the correct state.

emanuelelaface, 2020-12-25:
Is it normal that the discovery is completely erratic? Somtime it does not discover it for hours, after many reboots etc, some other times it find it in couple of seconds and works fine. After the discovery it always work fine but the discovery process seems really random.
By the way, I was monitoring all the times in Bonjour Explorer (now Discovery) and I can see it from the boot the announcements, I can see them passing also on tcpdump. It seems an iPhone issue but other devices are always discovered at the first shot.