espressif/esp-apple-homekit-adk #22: Make it easier to add component to project

slompf18, 2021-01-06:
Adding the component as submodule in components directory of a project fails, until you add

`set(EXTRA_COMPONENT_DIRS ${CMAKE_CURRENT_LIST_DIR}/components/esp-apple-homekit-adk)`

 to your CMakeLists.txt. I guess its because the CMakeLists.txt of the component is not in the root directory as stated [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#component-cmakelists-files).

Is this an intended behavior? Otherwise I could fix it.

shahpiyushv, 2021-01-07:
@slompf18 , there's no specific reason why CMakeLists.txt isn't in the root directory here. It's just that some of our other recent projects had multiple components in them and so we did not have the CMakeLists.txt for them in the root folder. The same logic got applied here, even though here it is a single component :)

slompf18, 2021-01-07:
Should we correct it? It's a fast fix and would make life a little bit easier for others using the component. Actually I have fixed it on a branch.