espressif/esp-apple-homekit-adk #24: error while compiling LightBulb Example

manvish, 2021-01-17:
[101/101] Generating binary image from built executable
esptool.py v3.0
Generated C:/esp/esp-idf/esp-apple-homekit-adk/examples/Lightbulb/build/bootloader/bootloader.bin
[996/1026] Building C object esp-idf/port/CMakeFiles/__idf_port.dir/__/homekit_adk/PAL/Crypto/MbedTLS/HAPMbedTLS.c.obj
FAILED: esp-idf/port/CMakeFiles/__idf_port.dir/__/homekit_adk/PAL/Crypto/MbedTLS/HAPMbedTLS.c.obj
ccache C:\esp\tools\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -DHAP_LOG_LEVEL=1 -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -Iconfig -I../../../port/include -I../../../homekit_adk/HAP -I../../../homekit_adk/PAL -I../../../homekit_adk/External/Base64 -I../../../homekit_adk/External/JSON -I../../../homekit_adk/External/HTTP -I../../../../components/newlib/platform_include -I../../../../components/freertos/include -I../../../../components/freertos/xtensa/include -I../../../../components/heap/include -I../../../../components/log/include -I../../../../components/lwip/include/apps -I../../../../components/lwip/include/apps/sntp -I../../../../components/lwip/lwip/src/include -I../../../../components/lwip/port/esp32/include -I../../../../components/lwip/port/esp32/include/arch -I../../../../components/soc/src/esp32/. -I../../../../components/soc/src/esp32/include -I../../../../components/soc/include -I../../../../components/esp_rom/include -I../../../../components/esp_common/include -I../../../../components/esp_system/include -I../../../../components/xtensa/include -I../../../../components/xtensa/esp32/include -I../../../../components/esp32/include -I../../../../components/driver/include -I../../../../components/driver/esp32/include -I../../../../components/esp_ringbuf/include -I../../../../components/efuse/include -I../../../../components/efuse/esp32/include -I../../../../components/espcoredump/include -I../../../../components/esp_timer/include -I../../../../components/esp_ipc/include -I../../../../components/soc/soc/esp32/include -I../../../../components/soc/soc/esp32/../include -I../../../../components/soc/soc/esp32/private_include -I../../../../components/vfs/include -I../../../../components/esp_wifi/include -I../../../../components/esp_wifi/esp32/include -I../../../../components/esp_event/include -I../../../../components/esp_netif/include -I../../../../components/esp_eth/include -I../../../../components/tcpip_adapter/include -I../../../../components/app_trace/include -I../../../../components/nvs_flash/include -I../../../../components/spi_flash/include -I../../../../components/mbedtls/port/include -I../../../../components/mbedtls/mbedtls/include -I../../../../components/mbedtls/esp_crt_bundle/include -I../../../../components/mdns/include -I../../../../components/console -mlongcalls -Wno-frame-address   -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.2\" -DESP_PLATFORM -MD -MT esp-idf/port/CMakeFiles/__idf_port.dir/__/homekit_adk/PAL/Crypto/MbedTLS/HAPMbedTLS.c.obj -MF esp-idf\port\CMakeFiles\__idf_port.dir\__\homekit_adk\PAL\Crypto\MbedTLS\HAPMbedTLS.c.obj.d -o esp-idf/port/CMakeFiles/__idf_port.dir/__/homekit_adk/PAL/Crypto/MbedTLS/HAPMbedTLS.c.obj   -c C:/esp/esp-idf/esp-apple-homekit-adk/homekit_adk/PAL/Crypto/MbedTLS/HAPMbedTLS.c
C:/esp/esp-idf/esp-apple-homekit-adk/homekit_adk/PAL/Crypto/MbedTLS/HAPMbedTLS.c: In function 'HAP_X25519_scalarmult':
C:/esp/esp-idf/esp-apple-homekit-adk/homekit_adk/PAL/Crypto/MbedTLS/HAPMbedTLS.c:119:19: error: implicit declaration of function 'mbedtls_ecp_read_key'; did you mean 'mbedtls_ecp_gen_key'? [-Werror=implicit-function-declaration]
         int ret = mbedtls_ecp_read_key(MBEDTLS_ECP_DP_CURVE25519, &our_key, n, X25519_SCALAR_BYTES);
                   ^~~~~~~~~~~~~~~~~~~~
C:/esp/esp-idf/esp-apple-homekit-adk/homekit_adk/PAL/Crypto/MbedTLS/HAPMbedTLS.c:107:9: note: in definition of macro 'WITH'
         X; \
         ^
C:/esp/esp-idf/esp-apple-homekit-adk/homekit_adk/PAL/Crypto/MbedTLS/HAPMbedTLS.c:118:5: note: in expansion of macro 'WITH_ECP_KEYPAIR'
     WITH_ECP_KEYPAIR(our_key, {
     ^~~~~~~~~~~~~~~~
cc1.exe: some warnings being treated as errors
[1001/1026] Building C object esp-idf/port/CMakeFiles/__idf_port.dir/__/homekit_adk/HAP/HAPIPAccessoryServer.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

manvish, 2021-01-17:
solved

closed 2021-01-17 by manvish

srleeb, 2021-02-21:
> solved

How? :-)

luisjimenez6245, 2021-03-29:
@srleeb You should pull the mbedtls 2.18 on you mbedtls path for making it work :)

$ cd $IDF_PATH/components/mbedtls/mbedtls
$ git pull
$ git checkout -b mbedtls-2.16.6-adk origin/mbedtls-2.16.6-adk