espressif/esp-apple-homekit-adk #25: New MbedTLS issue

nicksaintjohn, 2021-01-20:
I have been compiling fine for the past month or two but, having taken a break for a couple of weeks, I have just returned to find that nothing will compile any more. I can only imagine that maybe some update has broken something. I'm unable to even compile the vanilla Lightbulb example.

I receive the following error when attempting to compile: -

/Users/nicksaintjohn/esp/esp-apple-homekit-adk/homekit_adk/PAL/Crypto/MbedTLS/HAPMbedTLS.c:119:19: error: implicit declaration of function 'mbedtls_ecp_read_key'; did you mean 'mbedtls_ecp_gen_key'? [-Werror=implicit-function-declaration]
         int ret = mbedtls_ecp_read_key(MBEDTLS_ECP_DP_CURVE25519, &our_key, n, X25519_SCALAR_BYTES);
                   ^~~~~~~~~~~~~~~~~~~~
/Users/nicksaintjohn/esp/esp-apple-homekit-adk/homekit_adk/PAL/Crypto/MbedTLS/HAPMbedTLS.c:107:9: note: in definition of macro 'WITH'
         X; \
         ^
/Users/nicksaintjohn/esp/esp-apple-homekit-adk/homekit_adk/PAL/Crypto/MbedTLS/HAPMbedTLS.c:118:5: note: in expansion of macro 'WITH_ECP_KEYPAIR'
     WITH_ECP_KEYPAIR(our_key, {
     ^~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
[974/1025] Building C object esp-idf/port/CMakeFiles/__idf_port.dir/Users/nicksaintjohn/esp/esp-apple-homekit-adk/homekit_adk/HAP/HAPAccessoryValidation.c.obj

I've tried updating and switching to various versions of esp-idf, sometimes yielding different (now fixed) build errors, but when using the latest build (currently on esp-idf v4.3, esptool.py v3.0) I always get stuck at this same error.

I noticed something similar was reported a few months back (#3), but I'm presumably using a much later build, and therefor this should already be fixed!?

Any help or pointers would be greatly appreciated. I'm on MacOS 11.1 (Big Sur)

shahpiyushv, 2021-01-20:
@nicksaintjohn , can you double check if you have followed the steps [here](https://github.com/espressif/esp-apple-homekit-adk#set-up-mbedtls) for setting up mbedTLS?

nicksaintjohn, 2021-01-20:
Okay, I switched to that branch (was on HEAD previously), but upon doing a 'git pull' it says I'm already up to date... It does report some issues however, so I suspect that may be the problem.

```
nicksaintjohn@Nicks-MacBook-Air mbedtls % git branch -l
* (HEAD detached at 90f46c8b1)
  mbedtls-2.16.6-adk
  mbedtls-2.16.7-idf
nicksaintjohn@Nicks-MacBook-Air mbedtls % git switch mbedtls-2.16.6-adk
Previous HEAD position was 90f46c8b1 Fix `maybe-uninitialized` warnings resulting in build failure.
Switched to branch 'mbedtls-2.16.6-adk'
Your branch is up to date with 'origin/mbedtls-2.16.6-adk'.

```

Could you please suggest how to rectify / diagnose further? Please note, I was able to compile a couple of weeks ago, so I know it was correctly set up in the past. To my knowledge nothing has been changed (aside from a reboot), so I've no idea how anything has broken. Thanks

nicksaintjohn, 2021-01-20:
Ok, seems I've just recompiled, and now it's working again... I'm at a loss as to what happened to screw it up, but you led me to the solution, so many thanks!

JorgeBeserra, 2021-12-15:
Thanks...!!

After switch using this command solved, for me...!

git switch mbedtls-2.16.6-ad