espressif/esp-apple-homekit-adk #26: Controlling Multiple pins from single accessory

manvish, 2021-01-23:
I want to control multiple pins  (3 GPIO and 3 PWM) from iphone Home App.
I have successfully controlled one pin from <HandleLightBulbOnWrite> method but i dont know how to control multiple pins.
pls help me with this issue,

pedros89, 2021-07-05:
Hello, yes I am trying to do the same. I need to make the prototype and then enroll to MFi program for my RGB lamp
and tried with the fan example [esp-homekit-sdk ](https://github.com/espressif/esp-homekit-sdk) but with a multi-meter I noticed that there wasn't any voltage change on any of the GPIOs. Can I ask how did you make to control at least one pin? thank you.
My aim would be to have the SDK for RGB lamp and know which are the 3 PWM pins.

ShareCat, 2022-10-14:
You can modify this func : HandleLightBulbOnWrite()