espressif/esp-apple-homekit-adk #28: Getting Uncertified Accessory message when adding device

rajtrada, 2021-08-02:
Can we remove the "Uncertified Accessory" alert when adding a device to HomeKit?

AramVartanyan, 2021-08-02:
The only way to remove it is to use the MFi version of the SDK. To get it, you will need to be MFi certified and to contact Espressif team. 😉

shahpiyushv, 2021-08-02:
@rajtrada , what @AramVartanyan is right. You need to have MFI License and then you can get in touch with our sales team [here](https://www.espressif.com/en/products/sdks/esp-homekit-sdk) to get access to the MFi variant of the SDK. Once you have the access, you also need to follow the Apple guidelines for commercial certified HomeKit accessories and then that Uncertified prompt will be gone :)

rajtrada, 2021-08-02:
Thanks, @AramVartanyan, and @shahpiyushv for the detailed explanation.

closed 2021-08-02 by rajtrada