espressif/esp-apple-homekit-adk #29: Lightbulb example is not working

lorenzofiamingo, 2021-08-09:
Following instruction, when monitoring I get this output with an error:

```console
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6396
ho 0 tail 12 room 4
load:0x40078000,len:14252
load:0x40080400,len:3820
0x40080400: _init at ??:?

entry 0x40080698
I (29) boot: ESP-IDF v4.4-dev-2533-g83956ebba-dirty 2nd stage bootloader
I (29) boot: compile time 12:39:07
I (30) boot: chip revision: 1
I (34) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 4MB
I (55) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 sec_cert         unknown          3f 00 0000d000 00003000
I (78) boot:  1 nvs              WiFi data        01 02 00010000 00006000
I (86) boot:  2 otadata          OTA data         01 00 00016000 00002000
I (93) boot:  3 phy_init         RF data          01 01 00018000 00001000
I (101) boot:  4 ota_0            OTA app          00 10 00020000 00190000
I (108) boot:  5 ota_1            OTA app          00 11 001b0000 00190000
I (116) boot:  6 fctry            WiFi data        01 02 00340000 00006000
I (123) boot: End of partition table
I (128) boot_comm: chip revision: 1, min. application chip revision: 0
I (135) esp_image: segment 0: paddr=00020020 vaddr=3f400020 size=297b0h (169904) map
I (205) esp_image: segment 1: paddr=000497d8 vaddr=3ffb0000 size=03adch ( 15068) load
I (211) esp_image: segment 2: paddr=0004d2bc vaddr=40080000 size=02d5ch ( 11612) load
I (216) esp_image: segment 3: paddr=00050020 vaddr=400d0020 size=a3194h (668052) map
I (461) esp_image: segment 4: paddr=000f31bc vaddr=40082d5c size=11a10h ( 72208) load
I (491) esp_image: segment 5: paddr=00104bd4 vaddr=50000000 size=00010h (    16) load
I (501) boot: Loaded app from partition at offset 0x20000
I (501) boot: Disabling RNG early entropy source...
I (513) cpu_start: Pro cpu up.
I (514) cpu_start: Starting app cpu, entry point is 0x400811a4
0x400811a4: call_start_cpu1 at /Users/lorenzofiamingo/esp/esp-idf/components/esp_system/port/cpu_start.c:156

I (500) cpu_start: App cpu up.
I (530) cpu_start: Pro cpu start user code
I (530) cpu_start: cpu freq: 160000000
I (530) cpu_start: Application information:
I (534) cpu_start: Project name:     Lightbulb
I (539) cpu_start: App version:      94ed975
I (544) cpu_start: Compile time:     Aug  9 2021 12:39:01
I (551) cpu_start: ELF file SHA256:  069b5405d849e32c...
I (557) cpu_start: ESP-IDF:          v4.4-dev-2533-g83956ebba-dirty
I (564) heap_init: Initializing. RAM available for dynamic allocation:
I (571) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (577) heap_init: At 3FFC2358 len 0001DCA8 (119 KiB): DRAM
I (583) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (589) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (596) heap_init: At 4009476C len 0000B894 (46 KiB): IRAM
I (603) spi_flash: detected chip: generic
I (607) spi_flash: flash io: dio
I (612) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
1970-01-01'T'00:00:06'Z'        Default [com.apple.mfi.HomeKit.Platform:KeyValueStore] keyValueStore nvs Initialised
1970-01-01'T'00:00:06'Z'        Default [com.apple.mfi.HomeKit.Platform:KeyValueStore] keyValueStore fctry Initialised
I (756) wifi:wifi driver task: 3ffcb984, prio:23, stack:6656, core=0
I (756) system_api: Base MAC address is not set
I (756) system_api: read default base MAC address from EFUSE
I (786) wifi:wifi firmware version: d062fdb
I (786) wifi:wifi certification version: v7.0
I (786) wifi:config NVS flash: enabled
I (786) wifi:config nano formating: disabled
I (786) wifi:Init data frame dynamic rx buffer num: 32
I (796) wifi:Init management frame dynamic rx buffer num: 32
I (796) wifi:Init management short buffer num: 32
I (806) wifi:Init dynamic tx buffer num: 32
I (806) wifi:Init static rx buffer size: 1600
I (806) wifi:Init static rx buffer num: 10
I (816) wifi:Init dynamic rx buffer num: 32
I (816) wifi_init: rx ba win: 6
I (826) wifi_init: tcpip mbox: 32
I (826) wifi_init: udp mbox: 6
I (826) wifi_init: tcp mbox: 6
I (836) wifi_init: tcp tx win: 5744
I (836) wifi_init: tcp rx win: 5744
I (846) wifi_init: tcp mss: 1440
I (846) wifi_init: WiFi IRAM OP enabled
I (846) wifi_init: WiFi RX IRAM OP enabled
1970-01-01'T'00:00:06'Z'        Default [com.apple.mfi.HomeKit.Platform:KeyValueStore] Error (4354). Key 20 not found in KeyStore
1970-01-01'T'00:00:06'Z'        Default [com.apple.mfi.HomeKit.Platform:KeyValueStore] Error (4354). Key 21 not found in KeyStore
I (876) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (996) wifi:mode : sta (24:6f:28:79:64:e4)
I (996) wifi:enable tsf
I (996) wifi station: wifi_init_sta finished.
1970-01-01'T'00:00:06'Z'        Default [com.apple.mfi.HomeKit.Core:AccessoryServer] Version information:
libhap: Unknown
  - Version: Internal (Aug  9 2021 12:39:26) - compatibility version 7
Using platform: ESP32
  - Version: v4.4-dev-2533-g83956ebba-dirty (Aug  9 2021 12:39:17) - compatibility version 7
  - Available features:
    - Key-Value store
    - Accessory setup manager
    - Service discovery
1970-01-01'T'00:00:06'Z'        Default [com.apple.mfi.HomeKit.Platform:KeyValueStore] Error (4354). Key 00 not found in KeyStore
1970-01-01'T'00:00:06'Z'        Default [com.apple.mfi.HomeKit.Platform:Clock] Using 'clock_gettime' with 'CLOCK_MONOTONIC_RAW'.
1970-01-01'T'00:00:06'Z'        Error   [com.apple.mfi.HomeKit.Platform:Clock] clock_gettime failed: 22.
1970-01-01'T'00:00:06'Z'        Fault   fatal error - HAPPlatformClockGetCurrent @ /Users/lorenzofiamingo/Developer/esp-apple-homekit-adk/port/src/HAPPlatformClock.c:51

abort() was called at PC 0x400d7687 on core 0
0x400d7687: syscall_not_implemented_aborts at /Users/lorenzofiamingo/esp/esp-idf/components/newlib/syscalls.c:34

Backtrace:0x40081a03:0x3ffc7a800x4008850d:0x3ffc7aa0 0x4008efce:0x3ffc7ac0 0x400d7687:0x3ffc7b30 0x4015f41a:0x3ffc7b50 0x400ea7e9:0x3ffc7b70 0x400da130:0x3ffc7b90 0x400ea8fa:0x3ffc7bb0 0x400d9045:0x3ffc7be0 0x400d7c6b:0x3ffc7c20 0x4008b891:0x3ffc7c40 
0x40081a03: panic_abort at /Users/lorenzofiamingo/esp/esp-idf/components/esp_system/panic.c:391

0x4008850d: esp_system_abort at /Users/lorenzofiamingo/esp/esp-idf/components/esp_system/esp_system.c:129

0x4008efce: abort at /Users/lorenzofiamingo/esp/esp-idf/components/newlib/abort.c:46

0x400d7687: syscall_not_implemented_aborts at /Users/lorenzofiamingo/esp/esp-idf/components/newlib/syscalls.c:34

0x4015f41a: exit at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/exit.c:64

0x400ea7e9: HAPPlatformAbort at /Users/lorenzofiamingo/Developer/esp-apple-homekit-adk/port/src/HAPPlatformAbort.c:27

0x400da130: HAPFatalErrorInternal at /Users/lorenzofiamingo/Developer/esp-apple-homekit-adk/homekit_adk/PAL/HAPAssert.c:44

0x400ea8fa: HAPPlatformClockGetCurrent at /Users/lorenzofiamingo/Developer/esp-apple-homekit-adk/port/src/HAPPlatformClock.c:51 (discriminator 1)

0x400d9045: HAPPlatformRunLoopRun at /Users/lorenzofiamingo/Developer/esp-apple-homekit-adk/port/src/HAPPlatformRunLoop.c:609 (discriminator 1)

0x400d7c6b: main_task at /Users/lorenzofiamingo/Developer/esp-apple-homekit-adk/examples/Lightbulb/main/app_main.c:336 (discriminator 2)

0x4008b891: vPortTaskWrapper at /Users/lorenzofiamingo/esp/esp-idf/components/freertos/port/xtensa/port.c:159
```

thncode, 2021-09-18:
I do get the same fatal error around HAPPlatformClockGetCurrent

has anybody solved this?

ratsept, 2021-11-17:
I'm also getting this same error. Using esp32s2 target (Olimex ESP32-S2-DevKit-Lipo board). Examples of the other HomeKit SDK (esp-homekit-sdk) work just fine.

ShareCat, 2021-11-22:
I'm also getting this same error. Using esp32 target.

And here is the log:

ELF file SHA256: 693d925e95e0bf30

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6840
load:0x40078000,len:14836
load:0x40080400,len:3792
0x40080400: _init at ??:?

entry 0x40080694
I (27) boot: ESP-IDF v5.0-dev-382-g098dfe8a08-dirty 2nd stage bootloader
I (27) boot: compile time 10:55:59
I (28) boot: chip revision: 1
I (32) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 4MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 sec_cert         unknown          3f 06 0000d000 00003000
I (76) boot:  1 nvs              WiFi data        01 02 00010000 00006000
I (84) boot:  2 otadata          OTA data         01 00 00016000 00002000
I (91) boot:  3 phy_init         RF data          01 01 00018000 00001000
I (99) boot:  4 ota_0            OTA app          00 10 00020000 00190000
I (106) boot:  5 ota_1            OTA app          00 11 001b0000 00190000
I (114) boot:  6 fctry            WiFi data        01 02 00340000 00006000
I (121) boot: End of partition table
I (126) boot_comm: chip revision: 1, min. application chip revision: 0
I (133) esp_image: segment 0: paddr=00020020 vaddr=3f400020 size=29ad4h (170708) map
I (203) esp_image: segment 1: paddr=00049afc vaddr=3ffb0000 size=03b28h ( 15144) load
I (209) esp_image: segment 2: paddr=0004d62c vaddr=40080000 size=029ech ( 10732) load
I (214) esp_image: segment 3: paddr=00050020 vaddr=400d0020 size=a43b0h (672688) map
I (461) esp_image: segment 4: paddr=000f43d8 vaddr=400829ec size=11ee0h ( 73440) load
I (491) esp_image: segment 5: paddr=001062c0 vaddr=50000000 size=00010h (    16) load
I (502) boot: Loaded app from partition at offset 0x20000
I (502) boot: Disabling RNG early entropy source...
I (514) cpu_start: Pro cpu up.
I (514) cpu_start: Starting app cpu, entry point is 0x40081218
0x40081218: call_start_cpu1 at /home/user/esp/esp-idf/components/esp_system/port/cpu_start.c:164

I (501) cpu_start: App cpu up.
I (530) cpu_start: Pro cpu start user code
I (530) cpu_start: cpu freq: 160000000 Hz
I (530) cpu_start: Application information:
I (535) cpu_start: Project name:     Lightbulb
I (540) cpu_start: App version:      94ed975-dirty
I (546) cpu_start: Compile time:     Nov 22 2021 10:55:54
I (552) cpu_start: ELF file SHA256:  693d925e95e0bf30...
I (558) cpu_start: ESP-IDF:          v5.0-dev-382-g098dfe8a08-dirty
I (565) heap_init: Initializing. RAM available for dynamic allocation:
I (572) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (578) heap_init: At 3FFC2440 len 0001DBC0 (118 KiB): DRAM
I (584) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (591) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (597) heap_init: At 400948CC len 0000B734 (45 KiB): IRAM
I (605) spi_flash: detected chip: generic
I (608) spi_flash: flash io: dio
I (613) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
1970-01-01'T'00:00:04'Z'	Default	[com.apple.mfi.HomeKit.Platform:KeyValueStore] keyValueStore nvs Initialised
[... 44 lines cut ...]
abort() was called at PC 0x400d7acb on core 0
0x400d7acb: syscall_not_implemented_aborts at /home/user/esp/esp-idf/components/newlib/syscalls.c:35

Backtrace:0x40081c52:0x3ffc7c300x40088745:0x3ffc7c50 0x4008f3d6:0x3ffc7c70 0x400d7acb:0x3ffc7ce0 0x4016071a:0x3ffc7d00 0x400eac41:0x3ffc7d20 0x400da588:0x3ffc7d40 0x400ead52:0x3ffc7d60 0x400d949d:0x3ffc7d90 0x400d80af:0x3ffc7dd0 0x4008bb4d:0x3ffc7df0 
0x40081c52: panic_abort at /home/user/esp/esp-idf/components/esp_system/panic.c:403

0x40088745: esp_system_abort at /home/user/esp/esp-idf/components/esp_system/esp_system.c:123

0x4008f3d6: abort at /home/user/esp/esp-idf/components/newlib/abort.c:46

0x400d7acb: syscall_not_implemented_aborts at /home/user/esp/esp-idf/components/newlib/syscalls.c:35

0x4016071a: exit at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/exit.c:64

0x400eac41: HAPPlatformAbort at /home/user/homekit/esp-apple-homekit-adk/port/src/HAPPlatformAbort.c:27

0x400da588: HAPFatalErrorInternal at /home/user/homekit/esp-apple-homekit-adk/homekit_adk/PAL/HAPAssert.c:44

0x400ead52: HAPPlatformClockGetCurrent at /home/user/homekit/esp-apple-homekit-adk/port/src/HAPPlatformClock.c:51 (discriminator 1)

0x400d949d: HAPPlatformRunLoopRun at /home/user/homekit/esp-apple-homekit-adk/port/src/HAPPlatformRunLoop.c:609 (discriminator 1)

0x400d80af: main_task at /home/user/homekit/esp-apple-homekit-adk/examples/Lightbulb/build/../main/app_main.c:336 (discriminator 2)

0x4008bb4d: vPortTaskWrapper at /home/user/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:130

ShareCat, 2021-11-29:
```text
1970-01-01'T'00:00:04'Z'	Error	[com.apple.mfi.HomeKit.Platform:Clock] clock_gettime failed: 22.
1970-01-01'T'00:00:04'Z'	Fault	fatal error - HAPPlatformClockGetCurrent @ /home/user/homekit/esp-apple-homekit-adk/port/src/HAPPlatformClock.c:51

abort() was called at PC 0x400d7acb on core 0
0x400d7acb: syscall_not_implemented_aborts at /home/user/esp/esp-idf/components/newlib/syscalls.c:35
```

seems its a clock problem

ShareCat, 2021-11-30:
file: HAPPlatformClock.c

line46:
e = clock_gettime(CLOCK_MONOTONIC_RAW, &t);
modify:
e = clock_gettime(CLOCK_MONOTONIC, &t);

shahpiyushv, 2021-12-10:
@ShareCat , thanks for the suggestion. The same has now been changed in the code.

closed 2021-12-10 by shahpiyushv